pybamm-team/PyBaMM #4495: Doctest failing

brosaplanella, 2024-10-07:
The doctests are currently failing across branches. The error message is
```
Exception occurred:
  File "<template>", line 3, in top-level template code
TypeError: can only concatenate str (not "_StrPath") to str
The full traceback has been saved in /tmp/sphinx-err-b73gufkd.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at <https://github.com/sphinx-doc/sphinx/issues>. Thanks!
nox > Command sphinx-build -b html -W --keep-going . /home/runner/work/PyBaMM/PyBaMM/.nox/docs/bin/../tmp/html failed with exit code 2
nox > Session docs failed.
Error: Process completed with exit code 1.
```
but for a full log please see: https://github.com/pybamm-team/PyBaMM/actions/runs/11208039171/job/31151139329

kratman, 2024-10-07:
I cannot reproduce this on MacOS, so I suspect it is a Linux issue related to a new version of a docs library. I am going to look into it this morning to see if there is a quick fix

kratman, 2024-10-07:
I also did not re-produce this with a python3.11-slim image on docker. I am trying with ubuntu:latest

kratman, 2024-10-07:
Also did not reproduce it on ubnutu:latest (python 3.12)

PR #4498 (2024-10-07): Fix doc failures in notebooks
# Description

I could not reproduce this in docker or locally (MacOS), but I saw this fix on another project.

Fixes #4495 

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installing-and-using-pre-commit) for how to set this up to run automatically when committing locally, in just two lines of code)
- [x] All tests pass: `$ python run-tests.py --all` (or `$ no [... cut]