pybamm-team/PyBaMM #4496: [Bug]: IDAKLUSolver output variables

rtimms, 2024-10-07:
### PyBaMM Version

24.9

### Python Version

3.11

### Describe the bug

Running the same experiment but with cycles tagged differently can make the `IDAKLUSolver` return a solution without all the variables. 

@MarcBerliner @martinjrobins 

### Steps to Reproduce

```
model = pybamm.lithium_ion.SPM()
parameter_values = pybamm.ParameterValues("Chen2020")
solver = pybamm.IDAKLUSolver()

# works
experiment1 = pybamm.Experiment(
    [
        (
            "Rest for 1 hour",
            "Charge at C/5 until 4.2 V",
            "Hold at 4.2 V until C/20",
        ),
    ]
)
sim1 = pybamm.Simulation(
    model, parameter_values=parameter_values, experiment=experiment1, solver=solver
)
sim1.solve()
sim1.plot()

# fails
experiment2 = pybamm.Experiment(
    [
        ("Rest for 1 hour"),
        (
            "Charge at C/5 until 4.2 V",
            "Hold at 4.2 V until C/20",
        ),
    ]
)
sim2 = pybamm.Simulation(
    model, parameter_values=parameter_values, experiment=experiment2, solver=solver
)
sim2.solve()
sim2.plot()

```

### Relevant log output

KeyError: "Cannot process variable 'Negative particle surface concentration [mol.m-3]' as it was not part of the solve. Please re-run the solve with `output_variables` set to include this variable."

kratman, 2024-10-07:
@rtimms This one is a known issue: #4386

rtimms, 2024-10-07:
But here I am not using `output_variables`, so I think this is slightly different to #4386

kratman, 2024-10-07:
Hopefully it is fixed along with the other issue. I think @pipliggins was working on it

MarcBerliner, 2024-10-07:
@kratman I believe this is unrelated to https://github.com/pybamm-team/PyBaMM/issues/4386

@rtimms this is a bug, but I think we're seeing strange behavior here because the cell is at 100% SOC, and you're charging it in the second step, which instantaneously violates the voltage limit. I can look into this if it's blocking

rtimms, 2024-10-07:
Yeah, not a huge blocker. I also think that is what is happening, but it is handled more nicely by the other solvers. Just a random edge case I came across.

pipliggins, 2024-10-23:
I checked this with #4534 to confirm it's not an `output_variables` specific problem - it's not, but related. 

Having the voltage limit violation at the start of a cycle step results in the final solution containing an empty array in the middle of `all_ys`. The assumption leading to the output_variables error was that the only time an empty numpy array would be present in `all_ys` is if the solver isn’t returning the full state vector, as with IDAKLU(output_variables=true), so an false error is being flagged. I fixed that by checking for an empty `t` array too in `_update_variables()`, i.e.:
```
      for i, (model, ts, ys, inputs, var_pybamm) in enumerate(
          zip(self.all_models, self.all_ts, self.all_ys, self.all_inputs, vars_pybamm)
      ):
          if (
              ys.size == 0
              and var_pybamm.has_symbol_of_classes(
                  pybamm.expression_tree.state_vector.StateVector
              )
              and not ts.size == 0
          ):
```
But using this I get a different error related to the hermite interpolation. Turning off hermite interpolation in the solver (and with the output_variables fix) the example runs as expected. I suspect the interpolation issue is also being caused by an unexpected empty array in the state vector, but haven’t dug that far.

closed 2024-10-23 by MarcBerliner