pybamm-team/PyBaMM #4523: [Bug]: what is the compatible version between CUDA, pybamm and jax

lcwxz1989, 2024-10-18:
### PyBaMM Version

24.9

### Python Version

3.10

### Describe the bug

I have encountered a bug as on the user guide demo of PyBaMM show.  I can make jax use GPU on version 4.26 but can not on version 4.27, and if I install pybamm[jax], the jax version will upgrade to 4.27.How to use the GPU in a compatible version with pybamm, jax and cuda

Below is the PyBaMM manual
[image]
Info text below
```
An NVIDIA GPU may be present on this machine, but a CUDA-enabled jaxlib is not ins
```
Now I need to use PyBaMM to leverage GPU accelerate on CUDA dirver version 12.2
**Jax Can not use GPU**
[image]
[image]

**Jax Can use**
command: docker run --gpus all -it --rm  nvcr.io/nvidia/jax:24.04-py3
[image]

### Steps to Reproduce

docker run --gpus all -it --rm  nvcr.io/nvidia/jax:24.04-py3
# jax.devices will get gpu
pip install pybamm[jax]
# jax.devices will can not get gpu again

kratman, 2024-10-18:
@jsbrittain I am guessing this is a question for you

jsbrittain, 2024-10-18:
@lcwxz1989  PyBaMM comes with the default Jax (CPU) install by default. You can upgrade this to support cuda on your system (cuda-12 by the look of it in this case) by:
```bash
pip install --upgrade "jax[cuda12]"
```

If you now re-launch Python and `import pybamm` and `jax`, the GPU warning will no longer be displayed and `jax.devices()` should show cuda as an available device. Details on installing Jax with support for various device can be found here: https://jax.readthedocs.io/en/latest/installation.html.

agriyakhetarpal, 2024-10-18:
I'll convert this to a discussion since it doesn't seem like there is a bug on our end – more details can be shared there. A CUDA-enabled JAX distribution is required for GPU support.