pybamm-team/PyBaMM #4524: [Bug]: IDAKLU Solver incompatible with OKane2022 NE OCP Interpolation

tomjholland, 2024-10-18:
### PyBaMM Version

24.9

### Python Version

3.11

### Describe the bug

Casadi error is returned when using the IDAKLU solver with the `OKane2022` parameter set. Appears to be related to the interpolation of the negative electrode OCP, as it is searching for `libcasadi_interpolatnt_bspline.dylib` and setting the OCP to the functional form in `Chen2020` eliminates the error. 

This is a separate issue, but I will tag #4365 as another issue caused by the interpolation of discrete points of NE OCP in this parameter set.

### Steps to Reproduce

```python
exp = pybamm.Experiment([ "Discharge at 1C until 2.5 V"])
model = pybamm.lithium_ion.DFN()
params = pybamm.ParameterValues("OKane2022")
solver = pybamm.IDAKLUSolver()
sim = pybamm.Simulation(model, experiment=exp, parameter_values=params, solver=solver)
sim.solve()
```
Produces the Runtime error:
```
RuntimeError: .../casadi/core/casadi_os.cpp:166: Assertion "handle!=nullptr" failed:
PluginInterface::load_plugin: Cannot load shared library 'libcasadi_interpolant_bspline.dylib': 
   (
    Searched directories: 1. casadipath from GlobalOptions
                          2. CASADIPATH env var
                          3. PATH env var (Windows)
                          4. LD_LIBRARY_PATH env var (Linux)
                          5. DYLD_LIBRARY_PATH env var (osx)
    A library may be 'not found' even if the file exists:
          * library is not compatible (different compiler/bitness)
          * the dependencies are not found
   )
  Tried '' :
    Error code: dlopen(libcasadi_interpolant_bspline.dylib, 0x0005): tried: 'libcasadi_interpolant_bspline.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibcasadi_interpolant_bspline.dylib' (no such file), '/Users/tom/anaconda3/envs/pybamm/lib/python3.11/site-packages/pybamm/.dylibs/libcasadi_interpolant_bspline.dylib' (no such file), '/Users/tom/anaconda3/envs/pybamm/bin/../lib/libcasadi_interpolant_bspline.dylib' (no such file), '/usr/lib/libcasadi_interpolant_bspline.dylib' (no such file, not in dyld cache), 'libcasadi_interpolant_bspline.dylib' (no such file), '/usr/local/lib/libcasadi_interpolant_bspline.dylib' (no such file), '/usr/lib/libcasadi_interpolant_bspline.dylib' (no such file, not in dyld cache)
  Tried '.' :
    Error code: dlopen(./libcasadi_interpolant_bspline.dylib, 0x0005): tried: './libcasadi_interpolant_bspline.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS./libcasadi_interpolant_bspline.dylib' (no such file), '/Users/tom/anaconda3/envs/pybamm/lib/python3.11/site-packages/pybamm/.dylibs/./libcasadi_interpolant_bspline.dylib' (no such file), '/Users/tom/anaconda3/envs/pybamm/bin/../lib/./libcasadi_interpolant_bspline.dylib' (no such file), '/usr/lib/./libcasadi_interpolant_bspline.dylib' (no such file, not in dyld cache), './libcasadi_interpolant_bspline.dylib' (no such file), '/Users/tom/Library/CloudStorage/OneDrive-ImperialCollegeLondon/PhD Workspace/11 - PyBaMM DMA Analysis/pybamm-settings/libcasadi_interpolant_bspline.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/tom/Library/CloudStorage/OneDrive-ImperialCollegeLondon/PhD Workspace/11 - PyBaMM DMA Analysis/pybamm-settings/libcasadi_interpolant_bspline.dylib' (no such file), '/Users/tom/Library/CloudStorage/OneDrive-ImperialCollegeLondon/PhD Workspace/11 - PyBaMM DMA Analysis/pybamm-settings/libcasadi_interpolant_bspline.dylib' (no such file)
```

This can be overcome by changing the OCP function of the parameter set:
```python
exp = pybamm.Experiment([ "Discharge at 1C until 2.5 V"])
model = pybamm.lithium_ion.DFN()
params = pybamm.ParameterValues("OKane2022")
params["Negative electrode OCP [V]"] = pybamm.ParameterValues("Chen2020")['Negative electrode OCP [V]']
solver = pybamm.IDAKLUSolver()
sim = pybamm.Simulation(model, experiment=exp, parameter_values=params, solver=solver)
sim.solve()
```

MarcBerliner, 2024-10-18:
@tomjholland just double-checking – are you on PyBaMM v24.9 or develop? As a temporary workaround, you can change the OCP interpolant to `"linear"`. 

cc @kratman

kratman, 2024-10-18:
For Linux and Mac this fix from #4487 can resolve this issue. The change is needed inside pybamm's main `__init__.py`:
```
# Fix Casadi import
import os
import sysconfig
if sys.platform == "win32":
    os.environ["CASADIPATH"] = os.path.join(sysconfig.get_path('purelib'), 'casadi').replace("\\", "\\\\")
else:
    os.environ["CASADIPATH"] = os.path.join(sysconfig.get_path('purelib'), 'casadi')
```
This does not work on Windows, I am working on improving the Windows build to make it work there as well.

@tomjholland For tracking issues:
- Which OS are you using?
- Are you using 24.9 installed from pip or from source?

kratman, 2024-10-18:
I can reproduce this on MacOS with PyBaMM 24.9 on MacOS

kratman, 2024-10-18:
@tomjholland This works on my Mac as a workaround:
```python
import pybamm

# Fix Casadi import
import os
import sysconfig
os.environ["CASADIPATH"] = os.path.join(sysconfig.get_path('purelib'), 'casadi')

exp = pybamm.Experiment([ "Discharge at 1C until 2.5 V"])
model = pybamm.lithium_ion.DFN()
params = pybamm.ParameterValues("OKane2022")
solver = pybamm.IDAKLUSolver()
sim = pybamm.Simulation(model, experiment=exp, parameter_values=params, solver=solver)
sim.solve()
```

tomjholland, 2024-10-18:
Hi @kratman, thanks for the quick response on this.

Sorry, I forgot to add this detail to the issue. I get this error from 24.9 installed from pip on MacOS and Linux. I can confirm that your workaround also works for me on my Mac.

Thanks for the help!

closed 2024-10-22 by kratman