pybamm-team/PyBaMM #4560: Refactor "Inner SEI lithium interstitial diffusivity [m2.s-1]" in Arrhenius form

Mingzefei, 2024-11-03:
### Description

I want to refactor "Inner SEI lithium interstitial diffusivity [m2.s-1]" in Arrhenius form. I have found the reference value for the activation energy of this reaction and intend to refactor this param as a function of temperature.

kratman, 2024-11-21:
@Mingzefei There are a few PRs working on SEI and other degradation mechanisms at the moment. You might want to start by taking a look at some of them to see how your idea fits into the current progress:
- https://github.com/pybamm-team/PyBaMM/pull/4470
- https://github.com/pybamm-team/PyBaMM/pull/4463
- https://github.com/pybamm-team/PyBaMM/pull/4417

These three PRs are all linked to some of the current degradation work. Once you see what is changing, then you can start to refactor and put up a PR.

PyBaMM also has a slack channel and some other forums where you could discuss it further, if needed

DrSOKane, 2024-11-25:
Hi Zefei, all SEI growth mechanisms have the same Arrhenius dependence, using the parameter `"SEI growth activation energy [J.mol-1]"`. So the feature you are requesting has already been implemented!

closed 2024-11-25 by DrSOKane