pybamm-team/PyBaMM #4561: [Bug]: Binary operators on concatenations assume that each concatenation has the same number of orphans

aabills, 2024-11-04:
### PyBaMM Version

develop

### Python Version

3.12

### Describe the bug

If a binary operator, such as `+`, is invoked with 2 concatenations (i.e. `isinstance(left, pybamm.Concatenation) and isinstance(right, pybamm.Concatenation) == True`), and the concatenations have a different number of orphans, the "extra orphans" are dropped. This can be easily seen [here](https://github.com/aabills/PyBaMM/blob/8ce4a20090fee6bae2251d54fd28b4b0d49abdfb/src/pybamm/expression_tree/binary_operators.py#L859).

This is causing reaction heating to not be present in DFN half-cell models (see comments in #4557)

### Steps to Reproduce

no MWE yet, but should be self evident in the code.

aabills, 2024-11-04:
I think there are two different options for "solving" this:

1) Raise an error when this situation occurs
2) Pass the identity of the "unmatched orphans".

I implemented (1) to check how impactful it will be, and the only test that fails is the specific case I found this bug in: Half-cell DFN models. Therefore, I am of the opinion that we should implement (1) and fix the underlying issue that causes it in the half-cell case.

rtimms, 2024-11-04:
On board with option 1.

PR #4562 (2024-11-12): Throw an error when concatenations have different number of children
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #4561 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functi [... cut]
closed 2024-11-12 by kratman