pybamm-team/PyBaMM #4577: Adding new ECM battery parameter set however we are missing information

nikiexpo, 2024-11-12:
I am working on converting a Battery database with some of the parameters for about 40 different batteries (ECM). However, the data collection was done by someone else and the information some of the value fields in the[ example ](https://github.com/pybamm-team/PyBaMM/blob/develop/src/pybamm/input/parameters/ecm/example_set.py) is missing like;

- Cell-jig heat transfer coefficient
- Jig thermal mass
- Jig-air heat transfer coefficient
- Entropic change
- RCR lookup limit

The database was initially intended to be used for Optimal Control Problems that involves batteries and not for Pybamm, I believe that's why a lot of the thermal properties are missing. **My question is if the dataset we have would be of any use to Pybamm and if there's a way around the missing thermal property information.**

valentinsulzer, 2024-12-12:
Hi @nikiexpo , sorry for the slow reply on this! This dataset would be super useful and we can definitely help you include it.
- The RCR parameter looks like it's unused so we should remove it
- The entropic change parameter can be set to zero
- We can add different thermal models so you don't need to define those parameters. Are you using a simple lumped thermal model? Or no thermal model?

nikiexpo, 2024-12-15:
Hi @valentinsulzer , thank you for your reply. I am more of a optimal control guy and not a batteries guy so not sure of the battery lingo but we use the following dynamics for the battery:

$$ \dot{x}_1 = \frac{u(t)}{Q} $$

```math  
\dot{x}_2 = \frac{-x_2(t)}{R_1 C_1} + \frac{u_1(t)}{C_1} 
```
```math
\dot{x}_3 = \frac{1}{m c_p} (R_0 u^2(t) - h A (x_3 - T_{amb})
```
where $u(t)$ is the input current and three states are State of Charge (SoC), Voltage over RC pair ($V_RC$) and Temperature. This was the model the data was intended to be used with. Its simple as the original aim was to only do Optimal Control with the dataset (like Optimal Fast Charging) but we now would like it to make it more open access and accessible. 

I wrote a model using the `BaseModel` class in your package but that only exposes input current as the mode of control like in the dynamics but we would like more flexibility so our dataset is useful to a lot more users. Please advice on how to proceed.

valentinsulzer, 2024-12-16:
Got it - this is a lumped thermal model

We've now made the `BaseModel` class compatible with experiments. As long as you define the "Voltage [V]" and "Battery voltage [V]" as variables (just set battery voltage equal to voltage), it should work just like any other PyBaMM model. You can post the model here if you need help

kratman, 2024-12-23:
@nikiexpo Did this solve your issues?

nikiexpo, 2025-01-01:
Hi @kratman, sorry for the late response I have not been working as I am on holiday, you can close this issue for now. I will reopen when I complete the model and parameter set or if I have any other question related to this matter.

closed 2025-01-01 by agriyakhetarpal

valentinsulzer, 2025-01-14:
@nikiexpo here is your model in a format that works with experiments, it did take quite a lot of wrangling to get it to work properly because of the way experiments are set up. I just picked some random parameter values to check it works.

Feel free to use this directly or adapt however you need. Let me know if you need help with this, having the ECM parameter sets available would be amazing!

```python
import numpy as np
import pybamm

# SoC,OCV [V]
ocv_data = np.array(
    [
        [-0.05, 2.5554448268104863],
        [-0.04, 2.6965888919665],
        [-0.03, 2.8354243224771927],
        [-0.019999999999999997, 2.9688823518713066],
        [-0.009999999999999995, 3.0922350590723373],
        [6.938893903907228e-18, 3.2],
        [0.010000000000000009, 3.287757008219401],
        [0.020000000000000004, 3.3538087399146304],
        [0.030000000000000013, 3.399652870803843],
        [0.04000000000000002, 3.4291922391832723],
        [0.05000000000000002, 3.447386944382996],
        [0.06000000000000001, 3.4590176950801275],
        [0.07000000000000002, 3.4678233620936925],
        [0.08000000000000003, 3.476084048710994],
        [0.09000000000000001, 3.484703007570683],
        [0.10000000000000002, 3.493689095797286],
        [0.11000000000000003, 3.502728275706306],
        [0.12000000000000004, 3.511554074568407],
        [0.13000000000000006, 3.52005791717772],
        [0.14, 3.5282594966721414],
        [0.15000000000000002, 3.5362495945389743],
        [0.16000000000000003, 3.54414179665201],
        [0.17000000000000004, 3.552029381461937],
        [0.18000000000000005, 3.5599442209911114],
        [0.19000000000000006, 3.567823637742663],
        [0.20000000000000007, 3.575499973015437],
        [0.21000000000000008, 3.5827344436581194],
        [0.22000000000000008, 3.5893034627842035],
        [0.23000000000000004, 3.595101950719793],
        [0.24000000000000005, 3.600193518404891],
        [0.25000000000000006, 3.604767946222255],
        [0.26000000000000006, 3.609043147354448],
        [0.2700000000000001, 3.6131853781770666],
        [0.2800000000000001, 3.6172840846165255],
        [0.2900000000000001, 3.6213636610560163],
        [0.3000000000000001, 3.625399887548962],
        [0.3100000000000001, 3.6293293141200134],
        [0.32000000000000006, 3.633062478797201],
        [0.33000000000000007, 3.636514052629894],
        [0.3400000000000001, 3.639642639606244],
        [0.3500000000000001, 3.642472401747954],
        [0.3600000000000001, 3.645077134196744],
        [0.3700000000000001, 3.6475402964452446],
        [0.3800000000000001, 3.649923562924022],
        [0.39000000000000007, 3.6522631255377758],
        [0.4000000000000001, 3.6545895448555097],
        [0.4100000000000001, 3.6569578058385055],
        [0.4200000000000001, 3.659475082751738],
        [0.4300000000000001, 3.6623081482556623],
        [0.4400000000000001, 3.6656474871617903],
        [0.4500000000000001, 3.6696287502556433],
        [0.46000000000000013, 3.6742607079162277],
        [0.47000000000000014, 3.6794229802710614],
        [0.48000000000000015, 3.6849392547522477],
        [0.49000000000000016, 3.6906629553905184],
        [0.5000000000000001, 3.696514081906836],
        [0.51, 3.702464088074995],
        [0.52, 3.708507645219953],
        [0.53, 3.7146559216846993],
[... 178 lines cut ...]
        "Heat transfer coefficient [W.m-2.K-1]": 100,
        "Surface area [m2]": 1,
        "Mass [kg]": 1,
        "Specific heat capacity [J.kg-1.K-1]": 1000,
        "R0 [Ohm]": 0.02,
        "R1 [Ohm]": 0.02,
        "C1 [F]": 1,
        "Current function [A]": 1,
        # These parameters are not used in the model, but are required by PyBaMM
        # because of the way experiments are defined
        "Number of electrodes connected in parallel to make a cell": 1,
        "Number of cells connected in series to make a battery": 1,
        "Electrode width [m]": 1,
        "Electrode height [m]": 1,
    }
)

experiment = pybamm.Experiment(
    ["Discharge at 1C until 2.5V", "Charge at 1C until 4.2V", "Hold at 4.2V until C/20"]
)
sim = pybamm.Simulation(model, parameter_values=parameter_values, experiment=experiment)
sim.solve(calc_esoh=False)

sim.plot(["Voltage [V]", "Current [A]", "Temperature [K]"])
```