pybamm-team/PyBaMM #4578: Updating parameters without default values

isaacbasil, 2024-11-12:
Hi, 

As per my previous issue posted today, I want to start contributing to Pybamm, but I have never contributed to an open-source project before, so I am looking for an easy entry point to understand the process. 

A second point came up in some work I was doing today: when trying to update a parameter that doesn't have a default value, the following error is given:
` "Cannot update parameter '{name}' as it does not have a default value. ({err.args[0]}). If you are sure you want to update this parameter, use param.update({{name: value}}, check_already_exists=False)"
                  `

However, if I follow this advice and use param.update({{name: value}} where param is a ParameterValues object, I get the following error, 
TypeError: unhashable type: 'dict'

Perhaps this is just a typo? If I remove one set of curly brackets it works fine - param.update({name: value}, check_already_exists=False)

Again, just looking for a simple fix to make my first pull request. Please let me know what you think of this and if I have missed something.

kratman, 2024-11-12:
@isaacbasil Yes, this is likely just a string parsing error. I will investigate it and make a fix

isaacbasil, 2024-11-12:
Ah ok. If it's as simple as what I described in the issue, could I try to make the fix? So that I can get experience contributing.

kratman, 2024-11-12:
> Ah ok. If it's as simple as what I described in the issue, could I try to make the fix? So that I can get experience contributing.

Sorry, I already had the fix done in another branch before I saw your comment

isaacbasil, 2024-11-13:
No problem

PR #4581 (2024-11-12): Misc. bug fixes
# Description

A couple minor bug fixes before the release

Fixes #4580
Fixes #4578 

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installing-and-using-pre-commit) for how to set this up to run automatically when committing locally, in just two lines of code)
- [x] All tests pass: `$ python run-tests.py --all` (or `$ nox -s tests`)
- [x] The documentation builds: `$ python [... cut]