pybamm-team/PyBaMM #4590: Adding New interpolator to pybamm.Interpolant class.

martinjrobins, 2024-11-15:
### Discussed in https://github.com/pybamm-team/PyBaMM/discussions/4589

<div type='discussions-op-text'>

<sup>Originally posted by **MartinFuentetaja** November 15, 2024</sup>
Hello everyone! I am trying to use Akima1D interpolator to interpolate the OCPs of gr/LFP cell. I have tried to use pchip instead of cubic spline (due to the flatness of the OCP of LFP). However, the following error arises:

NotImplementedError: The interpolator 'pchip' is not supported by CasAdi. Use 'linear' or 'cubic' instead. Alternatively, set 'model.convert_to_format = 'python'' and use a non-CasADi solver. 

Therefore, I have also tried to implement Akima1D interpolator from scratch. However, I did not success. 

I have to mention that it works with linear interpolator, however I would prefer to use another one.</div>

martinjrobins, 2024-11-15:
### Discussed in https://github.com/pybamm-team/PyBaMM/discussions/4589

This is possible, the main piece of work would be to implement the pchip interpolant in the file pybamm/expression_tree/operations/convert_to_casadi.py. There is an issue on the casadi repo that gives an example of how to do this: https://github.com/casadi/casadi/issues/2489

Rishab87, 2025-02-23:
Hi @martinjrobins , I would like to work on this issue, can this be assigned to me?

PR #4871 (2025-02-24): Added pchip Interpolator
# Description
Added pchip Interpolator support in PyBaMM

Fixes #4590 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #)

# Important checks:

Please confirm the following before marking the PR as ready for review:
- [x] No style issues: `nox -s pre-commit`
- [x] All tests pass: `nox -s tests`
- [x] The documentation builds: `nox -s doctests`
- [x] Code is commented for hard-to-understand areas
- [x] Tests added that prove fix is effectiv [... cut]
closed 2025-02-24 by MarcBerliner