pybamm-team/PyBaMM #4611: Mark JAX IDAKLU solver as experimental on Windows

rtimms, 2024-11-25:
@martinjrobins @kratman can I leave you to fill in details, please?

kratman, 2024-11-25:
Yep, I will fill it in and update it this week

kratman, 2024-11-25:
I will include this in #4487 since it blocks that PR

martinjrobins, 2024-12-17:
I think the TODOs on this one are:
- [ ] skip JAX IDAKLU tests on windows (or run them but ignore the result)
- [ ] put some more details in the docs here: https://docs.pybamm.org/en/stable/source/user_guide/installation/gnu-linux-mac.html#optional-jaxsolver. This should probably list the different jax solvers (IDAKLU JAX and pure JAX), and state that IDAKLU JAX is experimental on windows
- [ ] add a note to the API docs here: https://docs.pybamm.org/en/stable/source/api/solvers/idaklu_jax.html, stating that this solver is experimental on windows

@kratman let me know if you want me to take on this one

kratman, 2024-12-17:
@martinjrobins Sorry for the delay on this one. I got a bit busy and was not able to complete it on time. The changes are fairly simple so I just need to sit down to do it

I am going to finish this over Christmas break so we can work on getting the IDAKLU separation into v25.1.0

PR #4487 (2025-01-07): Download IDAKLU from pybammsolvers
# Description

This will separate the IDAKLU C++ code from pybamm.

Fixes #3564
Fixes #4611

This ticket is something that can be handled in the other repo
Closes #3603 

## Type of change

This should speed up CI by skipping the build of the C++ code.

- [x] Optimization (back-end change that speeds up the code)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installing-and-using-pre-commit) for how to set this up to run automatic [... cut]
closed 2025-01-07 by kratman