pybamm-team/PyBaMM #4620: [Bug]: Variables names missing from plot

rtimms, 2024-11-26:
### PyBaMM Version

24.11.1

### Python Version

3.11

### Describe the bug

Dynamic plot has no variable labels and shows the legend entries as `t=0.00h` for all plots

[image]

### Steps to Reproduce

```python3
import pybamm

model = pybamm.lithium_ion.SPM()

sim = pybamm.Simulation(model)
sim.solve([0, 3600])

sim.plot()
```

### Relevant log output

```shell

```

rtimms, 2024-11-26:
From https://github.com/pybamm-team/PyBaMM/pull/4529

kratman, 2024-11-26:
@rtimms Sorry missed in the review

agriyakhetarpal, 2024-11-26:
cc: @medha-14, this might be worth a look!

brosaplanella, 2024-11-27:
Another issue that I believe is related: when plotting 2D variables the colourbar range is wrong:

## 24.9
[image]

## 24.11
[image]

brosaplanella, 2024-11-27:
Digging a bit deeper, there also some inconsistencies when plotting multiple solutions in one plot. Given that there are multiple issues to fix (basically the issue is a lot harder than I anticipated in #4271 😅) and plotting is quite a major feature, I suggest reverting #4529 and making a minor release. We can then address this for the next major release.

[image]

PR #4622 (2024-11-27): Revert "Modifies `QuickPlot.plot` ..."
# Description

This reverts commit b50d4a95c2639a4e10644332145d33d14e7b7594.

Fixes #4620

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installing-and-using-pre-commit) for how to set this up to run automatically when committing locally, in just two lines of code)
- [x] All tests pass: `$ python run-tests.py --all` (or `$ nox -s tests`)
- [x] The doc [... cut]
closed 2024-11-27 by kratman