pybamm-team/PyBaMM #4631: Review dependencies in documentation

Saransh-cpp, 2024-11-29:
I can see that a few PyBaMM dependencies are missing from - https://docs.pybamm.org/en/stable/source/user_guide/installation/index.html.

The dependencies in `pyproject.toml` and documentation should match.

agriyakhetarpal, 2024-11-29:
@prady0t, Saransh is conducting a hackathon next week. Could we please leave some of the new issues he's opening and the existing low-priority ones for newcomers? Thanks!

agriyakhetarpal, 2024-11-29:
@prady0t, Saransh is conducting a hackathon at UCL next week. Could we please leave some of the new issues he's opening and the existing low-priority ones for newcomers? Thanks!

prady0t, 2024-11-29:
Oh I wasn't aware thanks for letting me know!

Saransh-cpp, 2024-12-10:
Feel free to take this up now, @prady0t :)

PR #4689 (2024-12-18): Adding list of missing dependencies to docs
# Description

Adding the missing dependencies to docs.

Fixes #4631
closed 2024-12-18 by kratman