pybamm-team/PyBaMM #4637: [Bug]: post-processing variables require pybamm.Symbol to have a mesh

martinjrobins, 2024-12-02:
### PyBaMM Version

develop

### Python Version

3.10

### Describe the bug

Adding a variable to a pre-built pybamm model causes a failure in `pybamm.solvers.processed_variable.py::process_variable()` because it doesn't have a `mesh` attribute

### Steps to Reproduce

1. create a pybamm model (e.g. SPM)
2. manually add a new variable (model.variables.update({"name": expression}))
3. solve the model then try and access the variable (sol["name"])

### Relevant log output

```shell

```

martinjrobins, 2024-12-02:
this doesn't normally pop up as we seem to add mesh somewhere in the pipeline, but it seems to occur if you add the new variable *after* the model is discretised

martinjrobins, 2024-12-02:
I think the main issue with our current approach is that `mesh` is not an attribute of the `Symbol` class, its just something that gets added somewhere along the line, which is not good. In general all class attribute should be created in the `__init__`, not afterwards. I would suggest just adding `self.mesh = None` in `Symbol.__init__`

rtimms, 2024-12-02:
Got it, makes sense. There are a lot of places where attributes get added outside of the `__init__`…

martinjrobins, 2024-12-03:
> Got it, makes sense. There are a lot of places where attributes get added outside of the `__init__`…

I know :( but this was a particularly blatent example that I thought we should clean up

rtimms, 2024-12-03:
Agreed!

PR #4644 (2024-12-03): bug: make sure symbol always has required mesh attribute
# Description

Fixes #4637

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installing [... cut]
closed 2024-12-03 by rtimms