pybamm-team/PyBaMM #4647: Autoresponse in Github discussions pointing people to Discourse

brosaplanella, 2024-12-05:
Now that we have migrated to Discourse, it would be useful to set up an autoresponse to newly created GitHub discussions pointing people to Discourse.

This action should do the job: https://github.com/marketplace/actions/discussion-auto-responder

agriyakhetarpal, 2024-12-05:
Also, to take up in the same PR: add links to Discourse in a more prominent place in the repository README + docs pages

Saransh-cpp, 2024-12-05:
Maybe also as an issue type -

[image]

agriyakhetarpal, 2024-12-18:
It looks like a lot of people are still posting in GitHub Discussions and might not be aware of Discourse at all, so I'll mark it with a higher priority for anyone who would like to take it up

agriyakhetarpal, 2024-12-18:
Of course, please feel free to. We should get this done quickly. Could you also test on your fork while you're at it?

ahamed24, 2024-12-22:
@arjxn-py if you dont mind. Could i take this up :)

arjxn-py, 2024-12-22:
Hi @ahamed24, yes if you want to give it a try, you're welcome

PR #4708 (2024-12-23): Added GitHub Discussion Autoresponder and Updated Links to Discourse in Docs and README
# GitHub Discussion Autoresponder Integration

# Description

This PR introduces a commit that adds an autoresponder to GitHub discussions, which redirects users to the [Discourse forum](https://pybamm.discourse.group/). The changes include the addition of the discussion_responder.yml GitHub Action to manage automatic responses to new discussions, updates to the documentation to guide users toward the Discourse forum for further discussion, and the inclusion of a link to the Discourse forum in the README.md [... cut]
closed 2024-12-23 by arjxn-py