pybamm-team/PyBaMM #4662: [Bug]: model.set_initial_conditions_from assumes state variable names consistent with output variables

martinjrobins, 2024-12-11:
### PyBaMM Version

develop

### Python Version

n/a

### Describe the bug

the `BaseModel.set_initial_conditions` function extracts the ic values from a `pybamm.Solution` object. It looks in the output variables of the model to find each variable in turn, then concatenates a new state vector from the evaluation of those output variables.

This works fine with our library models, where for each variable `a = pybamm.Variable('A')`, we have defined an output variable `model.variables = { 'A': a}`. But third party models do not neccessarily have this structure, so the   `BaseModel.set_initial_conditions` function can easily fail when this third party model is used in an experiment

Perhaps instead we could:
1. disallow any output variables of a model with the same name as a model variable
2. automatically create output variables for each variable when model is discretised

### Steps to Reproduce

1. create a model that has a variable with name `A`, and an output variable with the same name but a different shape
2. use this model in an experiment

### Relevant log output

```shell

```

TusharNaugain, 2024-12-12:
> ### PyBaMM Version
> develop
>
> ### Python Version
> n/a
>
> ### Describe the bug
> the `BaseModel.set_initial_conditions` function extracts the ic values from a `pybamm.Solution` object. It looks in the [...]

Hi team,
I would like to contribute to resolving this issue. The bug description is clear, and I believe the outlined potential solutions provide a good starting point for the investigation.

To ensure compatibility between model variables and output variables:

Disallowing duplicate names would enforce better structure and avoid conflicts.
Automatically creating output variables for model variables upon discretization seems like an elegant solution to ensure consistent behavior for both library and third-party models.
May I take ownership of this issue? Please assign it to me, and I will proceed with a detailed exploration and propose a fix.

Thank you!

agriyakhetarpal, 2024-12-12:
Hi @TusharNaugain, your potential solution seems to be just a rephrase of the steps that have been outlined already. Could you please elaborate on how you plan to implement these changes? A proposal to fix this must be refined enough and filled with details that describe your approach to the codebase succinctly. 

This might not be a "good first issue", since it assumes some knowledge of what is going on with the processing and solving of our models – please look for issues labelled as "easy" if you wish to contribute and are looking for where to start!

TusharNaugain, 2024-12-13:
Hi @agriyakhetarpal ,

Thank you for your feedback. I apologize for the brevity of my initial response. Here’s a more detailed proposal for fixing the set_initial_conditions issue:

Detailed Proposal
Problem: The set_initial_conditions function can fail when third-party models have naming conflicts between model variables and output variables.

Solution:

Check for Name Clashes:

Implementation: Add a check in set_initial_conditions to compare model variable names with output variable names. Raise an error if a conflict is detected.

def set_initial_conditions(self, variables):
     Ensure no name clashes between model variables and output variables
    variable_names = set(variables.keys())
    output_variable_names = set(self.output_variables.keys())
    
    common_names = variable_names.intersection(output_variable_names)
    if common_names:
        raise ValueError(f"Variable names clash with output variables: {common_names}")

     Existing logic to set initial conditions
   
    
Testing:

Create Test Cases: Validate changes with test cases for models with potential naming conflicts.

Run Models: Ensure models process and solve without errors.

Compatibility: Verify that existing functionality remains unaffected.

Additional Notes: These changes will improve compatibility with third-party models and enhance error handling for a smoother user experience.

I understand this might not be a "good first issue" due to its complexity. Thank you for your guidance. I’ll look for issues labeled "easy" to start contributing effectively.

Thank you for your support and understanding.

Best regards, TusharNaugain

martinjrobins, 2024-12-13:
agree with @agriyakhetarpal that this is not a good first issue, you need to have a good knowledge of how pybamm represents and processes a model via discretisation.

I didn't implement the `BaseModel.set_initial_conditions` function, @valentinsulzer I think this was you, are you happy with my proposed solution?

valentinsulzer, 2024-12-19:
I think adding the variables when discretizing is a good idea, not sure why we also need to disallow outputs with the same name  as `set_initial_conditions` should just look for the variable with the right name and ignore any others

martinjrobins, 2024-12-19:
if there is already an output with the same name as the variable then the discretisation class won't be able to add the variable (its a dict mapping str to an expression as I recall)

PR #4700 (2025-01-08): feat: auto-add state variables to output variables
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #4662

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [x] Bug  [... cut]
closed 2025-01-08 by kratman