pybamm-team/PyBaMM #4678: Error _“The linesearch algorithm failed with too small a step.”_ in sensitivitiy analysis.

Azure-Dreamer, 2024-12-15:
Thank you pybamm team for your great work, and I would be grateful if you can provide some help with the following problem.

I tried to calculate the sensitivity applying the "explicit sensitivity calculation" function.

I created a new parameter class and put both negative and positive particle diffusivities into it, then processed it into InputParameter class during discretization . Everything is fine until solving. The solver usually raised error “The linesearch algorithm failed with too small a step.” when I chose "fast", "fast with events" and "safe" as the casadisolve "mode" argument. I tried

calculating the sensitiviy of only one parameter instead of both, for example by set calculate_sensitivities=['Negative particle diffusivity [m2.s-1]',],
different simulation time.
At present the only situation that solver didn't raise the error "The linesearch ..." is only sensitivity of Negative particle diffusivity is calculated and the simulation time is set as 0.1s. However, another error raised which seems not by solver:
[ERROR] solution.check_ys_are_not_too_large(397): Solution for '3.5986316539448007 + y[882:902]' exceeds the maximum allowed value of `100000.0. This could be due to incorrect scaling, model formulation, or parameter values. The maximum allowed value is set by 'pybammm.settings.max_y_value'.

I will post the error in the last of this issue, and thank you team for your work. If possible, please help me with it. I will also post it in [PyBaMM Discourse](https://pybamm.discourse.group/).

```python
The linesearch algorithm failed with too small a step.
2024-12-16 00:41:14.184 - [ERROR] callbacks.on_experiment_error(233): Simulation error: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:361:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.
Traceback (most recent call last):
File "F:\PyBaMM\pybamm\solvers\casadi_solver.py", line 685, in _run_integrator
casadi_sol = integrator(x0=y0_diff, z0=y0_alg, p=inputs_with_tmin, **self.extra_options_call) # self.extra_options_call: {}
File "F:\Anaconda\envs\PyBaMM_env\lib\site-packages\casadi\casadi.py", line 23531, in call
return self.call(kwargs)
File "F:\Anaconda\envs\PyBaMM_env\lib\site-packages\casadi\casadi.py", line 20180, in call
return _casadi.Function_call(self, *args)
RuntimeError: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:361:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
File "F:\PyBaMM\experiment.py", line 40, in
solution = sim.solve(inputs=inputs)
File "F:\PyBaMM\pybamm\simulation.py", line 778, in solve
raise error
File "F:\PyBaMM\pybamm\simulation.py", line 754, in solve
step_solution = solver.step(
File "F:\PyBaMM\pybamm\solvers\base_solver.py", line 1374, in step
solution = self._integrate(model, t_eval, model_inputs, t_interp)
File "F:\PyBaMM\pybamm\solvers\casadi_solver.py", line 169, in _integrate
solution = self._run_integrator(model, model.y0, inputs_dict, inputs, t_eval)
File "F:\PyBaMM\pybamm\solvers\casadi_solver.py", line 692, in _run_integrator
raise pybamm.SolverError(error.args[0]) from error
pybamm.expression_tree.exceptions.SolverError: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:361:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.

Process finished with exit code 1

Azure-Dreamer, 2024-12-15:
I also tried the .factory function of casadi to calculate the sensitivity, similar to casadi/casadi#2517.

briefly
```python
integrator_fwd_p = integrator.factory("F_fwd", ['x0', 'z0', 'p', 'fwd:p'], ['fwd:xf'])
casadi_sol_fwd_p = integrator_fwd_p(x0=y0_diff, z0=y0_alg, p=inputs_with_tmin, fwd_p=[1, 0, 0])
```

Similar error "The linesearch algorithm failed with too small a step." and another "Newton/Linesearch algorithm failed to converge." raised for "safe" and "fast" mode.

Following is the error when "safe" mode set.
```python
F:\Anaconda\envs\PyBaMM_env\python.exe F:\PyBaMM\experiment.py 
Newton/Linesearch algorithm failed to converge.
Function fsens1_F (0x20926946a60)
Input 0 (x0): 
[[0, 0], 
 [0, 0], 
 [0.81, 0], 
 ...,
 [0.426, 0], 
 [0.426, 0], 
 [0.426, 0]]
Input 1 (z0): 
[[-9.86656e-08, 0], 
 [-2.86168e-07, 0], 
 [-4.6384e-07, 0], 
 [-6.31682e-07, 0], 
 [-7.89691e-07, 0], 
 [-9.37865e-07, 0], 
 [-1.0762e-06, 0], 
 [-1.2047e-06, 0], 
 [-1.32335e-06, 0], 
 [-1.43215e-06, 0], 
 [-1.5311e-06, 0], 
 [-1.62019e-06, 0], 
 [-1.69941e-06, 0], 
 [-1.76876e-06, 0], 
 [-1.82823e-06, 0], 
 [-1.87781e-06, 0], 
 [-1.91749e-06, 0], 
 [-1.94727e-06, 0], 
 [-1.96713e-06, 0], 
 [-1.97707e-06, 0], 
 [-0.08144, 0], 
 [-0.0814562, 0], 
 [-0.0814883, 0], 
 [-0.0815361, 0], 
 [-0.0815997, 0], 
 [-0.0816789, 0], 
 [-0.0817738, 0], 
 [-0.0818844, 0], 
 [-0.0820108, 0], 
 [-0.0821533, 0], 
 [-0.082312, 0], 
 [-0.0824873, 0], 
 [-0.0826795, 0], 
 [-0.082889, 0], 
 [-0.0831164, 0], 
 [-0.083362, 0], 
 [-0.0836266, 0], 
 [-0.0839108, 0], 
 [-0.0842154, 0], 
 [-0.0845411, 0], 
 [-0.0744532, 0], 
 [-0.0744559, 0], 
 [-0.0744612, 0], 
 [-0.0744691, 0], 
 [-0.0744796, 0], 
 [-0.0744928, 0], 
[... 99 lines cut ...]
Error in Function::operator() for 'fsens1_F' [IdasInterface] at .../casadi/core/function.cpp:1513:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_CONV_FAIL". Consult IDAS documentation.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "F:\PyBaMM\experiment.py", line 40, in <module>
    solution = sim.solve(inputs=inputs)
  File "F:\PyBaMM\pybamm\simulation.py", line 778, in solve
    raise error
  File "F:\PyBaMM\pybamm\simulation.py", line 754, in solve
    step_solution = solver.step(
  File "F:\PyBaMM\pybamm\solvers\base_solver.py", line 1374, in step
    solution = self._integrate(model, t_eval, model_inputs, t_interp)
  File "F:\PyBaMM\pybamm\solvers\casadi_solver.py", line 169, in _integrate
    solution = self._run_integrator(model, model.y0, inputs_dict, inputs, t_eval)
  File "F:\PyBaMM\pybamm\solvers\casadi_solver.py", line 692, in _run_integrator
    raise pybamm.SolverError(error.args[0]) from error
pybamm.expression_tree.exceptions.SolverError: Error in Function::call for 'F_fwd' [MXFunction] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F_fwd' [MXFunction] at .../casadi/core/function.cpp:361:
Error in Function::operator() for 'fwd1_F' [MXFunction] at .../casadi/core/function.cpp:1513:
Error in Function::operator() for 'fsens1_F' [IdasInterface] at .../casadi/core/function.cpp:1513:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_CONV_FAIL". Consult IDAS documentation.

Process finished with exit code 1

Azure-Dreamer, 2024-12-15:
There is no problem when regular solving without sensitivity.

martinjrobins, 2024-12-17:
Hi @Azure-Dreamer, I'm unsure exactly what you are trying to do, can  you please post a MRE or small script that produces the error, so that I can run this locally and see what the problem is?

Azure-Dreamer, 2024-12-17:
@martinjrobins Thank you for your reply. A simple example for the first situation is like this:
```python
import pybamm
import numpy as np

experiment = pybamm.Experiment(["Discharge at 1 C for 10 seconds",])

parameter_values = pybamm.ParameterValues('Prada2013')
parameter_values["Negative electrode diffusivity [m2.s-1]"] = "[input]"

options = {}
model = pybamm.lithium_ion.DFN(options=options)

solver = pybamm.CasadiSolver()
sim = pybamm.Simulation(model, experiment=experiment, parameter_values=parameter_values, solver=solver)

inputs = {"Negative electrode diffusivity [m2.s-1]": 3e-15,}

t_eval = np.linspace(0, 30, 1000)
# sim.solve(t_eval=t_eval, inputs=inputs)  # this works for DAE case
sim.solve(t_eval=t_eval, inputs=inputs, calculate_sensitivities=True)
```
which outputs
```python
F:\Anaconda\envs\PyBaMM_env\python.exe F:\PyBaMM_2\experiment.py 
2024-12-18 00:11:13.787 - [WARNING] simulation.solve(556): Ignoring t_eval as solution times are specified by the experiment
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
2024-12-18 00:11:16.296 - [ERROR] callbacks.on_experiment_error(233): Simulation error: Maximum number of decreased steps occurred at t=0.0 (final SolverError: 'Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:361:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.'). For a full solution try reducing dt_max (currently, dt_max=600) and/or reducing the size of the time steps or period of the experiment. Set `return_solution_if_failed_early=True` to return the solution object up to the point where failure occured.
Traceback (most recent call last):
  File "F:\PyBaMM_2\pybamm\solvers\casadi_solver.py", line 678, in _run_integrator
    casadi_sol = integrator(
  File "F:\Anaconda\envs\PyBaMM_env\lib\site-packages\casadi\casadi.py", line 23531, in __call__
    return self.call(kwargs)
  File "F:\Anaconda\envs\PyBaMM_env\lib\site-packages\casadi\casadi.py", line 20180, in call
    return _casadi.Function_call(self, *args)
RuntimeError: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:361:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "F:\PyBaMM_2\pybamm\solvers\casadi_solver.py", line 245, in _integrate
    current_step_sol = self._run_integrator(
  File "F:\PyBaMM_2\pybamm\solvers\casadi_solver.py", line 684, in _run_integrator
    raise pybamm.SolverError(error.args[0]) from error
pybamm.expression_tree.exceptions.SolverError: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:361:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "F:\PyBaMM_2\experiment.py", line 70, in <module>
    sim.solve(t_eval=t_eval, inputs=inputs, calculate_sensitivities=True)
  File "F:\PyBaMM_2\pybamm\simulation.py", line 779, in solve
    raise error
  File "F:\PyBaMM_2\pybamm\simulation.py", line 755, in solve
    step_solution = solver.step(
  File "F:\PyBaMM_2\pybamm\solvers\base_solver.py", line 1364, in step
    solution = self._integrate(model, t_eval, model_inputs, t_interp)
  File "F:\PyBaMM_2\pybamm\solvers\casadi_solver.py", line 282, in _integrate
    raise pybamm.SolverError(
pybamm.expression_tree.exceptions.SolverError: Maximum number of decreased steps occurred at t=0.0 (final SolverError: 'Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:361:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.'). For a full solution try reducing dt_max (currently, dt_max=600) and/or reducing the size of the time steps or period of the experiment. Set `return_solution_if_failed_early=True` to return the solution object up to the point where failure occured.

Process finished with exit code 1
```

martinjrobins, 2024-12-17:
thanks @Azure-Dreamer. I would use the IDAKLU solver for this, I've tried this and it seemed to work. The sensitivity implementation for the IDAKLU solver uses the sundials library directly and is more reliable and faster than the casadi solver.

kratman, 2024-12-19:
@Azure-Dreamer The idaklu solver comes with the pypi installation. You can use a virtual environment to install pybamm:

Linux/Mac
```bash
virtualenv .venv
source .venv/bin/activate
pip install pybamm
```
That should let you use `IDAKLUSolver()` for your calculations