pybamm-team/PyBaMM #4720: Using `pytest-testmon` pluggin.

prady0t, 2024-12-27:
[pytest-testmon](https://testmon.org/) is a very useful pytest fixture. It runs only those tests where the related changes have been done. It can be helpful as an option for local development and CI as it would be significantly faster (since we are running a lot lesser tests).

prady0t, 2024-12-27:
Originally suggested by @agriyakhetarpal

kratman, 2024-12-27:
Something like that is fine for local dev, but we should test everything for CI. Dependencies can change, we have a complex codebase, and I am not sure anyone really looks at the periodic tests.

I don't think this is worth the effort to implement

agriyakhetarpal, 2024-12-27:
I agree with Eric. We would like to always test everything in CI, but please feel free to test it out over a few months on your fork and see how reliable it is.

In the meantime, documentation on how to set it up and run it so that someone can test the changes they've made in a local feature branch (maybe through a `nox` session or a positional argument to the `tests` session) would be helpful, such that it picks up the common ancestor between the default branch and said branch automatically.

kratman, 2024-12-27:
I think the other thing is that tests don't take that long locally. Running the unit tests takes about a minute. We don't need to go overboard with infrastructure or documentation

agriyakhetarpal, 2024-12-28:
Yeah, fair point – the integration tests have usually passed for me in two minutes, and I expect this to be the same on modern computers running Windows and Linux as well. Unless @prady0t has something else to suggest, we could close this.

prady0t, 2024-12-28:
Yes I guess it's not much needed then. This would have anyways made sense in local development but have never personally felt the need that much. Users already have an option to run selected tests via `nox -s tests -- "path"`. I guess we should close this.

agriyakhetarpal, 2024-12-28:
Closing as "not planned" so for tracking this better down the line

closed 2024-12-28 by agriyakhetarpal