pybamm-team/PyBaMM #4773: [Bug]: ec reaction sei growth model seems so different in 25.1 from previous versions

Brushnie, 2025-01-17:
### PyBaMM Version

25.1, 24.11.2

### Python Version

25.1, 24.11.2

### Describe the bug

I ran calendar_ageing.py from the example folder, the results for **"reaction limited"** sei model are the same for 25.1 and 24.11.2 versions, but when I changed the sei growth model to **"ec reaction limited"**, the results from version 25.1 are very different from 24.11.2.  @valentinsulzer 

[image]

[image]

### Steps to Reproduce

import pybamm as pb
import numpy as np

pb.set_logging_level("INFO")

models = [
    pb.lithium_ion.SPM({"SEI": "ec reaction limited"}),
    pb.lithium_ion.SPMe({"SEI": "ec reaction limited"}),
    pb.lithium_ion.SPM(
        {"SEI": "ec reaction limited", "surface form": "algebraic"},
        name="Algebraic SPM",
    ),
    pb.lithium_ion.SPMe(
        {"SEI": "ec reaction limited", "surface form": "algebraic"},
        name="Algebraic SPMe",
    ),
    pb.lithium_ion.DFN({"SEI": "ec reaction limited"}),
]

sims = []
for model in models:
    parameter_values = pb.ParameterValues("OKane2022") # model.default_parameter_values

    parameter_values["Current function [A]"] = 0

    sim = pb.Simulation(model, parameter_values=parameter_values)

    solver = pb.IDAKLUSolver()

    years = 0.50
    days = years * 365
    hours = days * 24
    minutes = hours * 60
    seconds = minutes * 60

    t_eval = np.linspace(0, seconds, 100)

    sim.solve(t_eval=t_eval, solver=solver)
    sims.append(sim)

pb.dynamic_plot(
    sims,
    [
        "Voltage [V]",
        "Negative particle surface concentration",
        "X-averaged negative particle surface concentration",
        "Electrolyte concentration [mol.m-3]",
        "Negative total SEI thickness [m]",
        "X-averaged negative total SEI thickness [m]",
        "X-averaged negative SEI concentration [mol.m-3]",
        "Sum of x-averaged negative electrode volumetric "
        "interfacial current densities [A.m-3]",
        "Loss of lithium inventory [%]",
        ["Total lithium lost [mol]", "Loss of lithium to negative SEI [mol]"],
    ],
)

### Relevant log output

```shell

```

aabills, 2025-01-17:
looking into it, there is for sure a bug here

aabills, 2025-01-17:
@Brushnie can you test #4774 and make sure it fixes your bug, and leave a comment there

Brushnie, 2025-01-17:
@aabills It seems to be fixed. Thanks!

aabills, 2025-01-17:
great, thanks for the report.

PR #4774 (2025-01-17): fix ec reaction bug
# Description

#4394 introduced a bug in ec-reaction limited sei, this reverts the problematic change

@kratman lets do a patch release
@kawaMANMI can you take a look at this and make sure it doesn't break your code. also please try and be careful with stuff like this

Fixes #4773 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-bre [... cut]