pybamm-team/PyBaMM #4776: [Bug]: DFN + MPM + stoichiometry-dependent diffusivity with string experiments

Daniel-Nicolae23, 2025-01-20:
### PyBaMM Version

25.1.0

### Python Version

3.12.7

### Describe the bug

Reviving bug #3549 

DFN + MPM + stoichiometry-dependent diffusivity works with a `pybamm.Experiment` containing only one string but fails when using multiple steps/cycles.

### Steps to Reproduce

```python
model = pybamm.lithium_ion.DFN(
    {
        "working electrode": "positive",
        "surface form": "differential",
        "contact resistance": "true",
        "particle size": "distribution",
    }
)

parameter_values = model.default_parameter_values

def f_a_dist_p_dim(R):
    return pybamm.lognormal(R, 8e-6, 1e-6)

distribution_params = {
    "Positive minimum particle radius [m]": 0,
    "Positive maximum particle radius [m]": 3e-5,
    "Positive area-weighted " + "particle-size distribution [m-1]": f_a_dist_p_dim,
}
parameter_values.update(distribution_params, check_already_exists=False)

def D_p(sto, T):
    return 8e-15 + 3e-15 * sto

parameter_values.update({"Positive particle diffusivity [m2.s-1]": D_p}, check_already_exists=False)

experiment = pybamm.Experiment(["Charge at C/100 for 20 hours", "Rest for 2 hours"]) # this fails
# experiment = pybamm.Experiment(["Charge at C/100 for 20 hours"]) # this is fine

sim = pybamm.Simulation(model, parameter_values=parameter_values, experiment=experiment, solver=pybamm.IDAKLUSolver())
sim.solve(initial_soc=0)
```

### Relevant log output

```shell
Shape not recognized for 48230.0 * y[2:12002](note processing of 3D variables is not yet implemented)
```

rtimms, 2025-01-20:
This will be the same issue as https://github.com/pybamm-team/PyBaMM/issues/1549

rtimms, 2025-01-20:
IMO we should do the second fix suggested below

> Yes, there are two possible fixes, either

edit processed variable so that 3D variables are allowed
or

add a hack to set_initial_conditions_from to avoid needing to call processed variable in 3D
The latter is probably easier. If the 3D variable is already a state in the system (which it will be in most cases), we can just directly call solution.y[slice] instead of solution[var.name]. The reason set_initial_conditions_from calls solution[var.name] by default is so you can initialize a model using a different model which does not have all the same states

kratman, 2025-01-20:
Part of #2394