pybamm-team/PyBaMM #4793: [Bug]: Error while modelling coupled degradation mechanism

gupspriya, 2025-01-22:
### PyBaMM Version

25.1.1

### Python Version

3.12

### Describe the bug

Hi, I am using PyBaMM for calculating LLI loss with respect to cycles. I am running for 1000 cycles but after around 600 cycles the model is giving below mentioned error:
At t = 100.79 and h = 2.04493e-10, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 243.462 and h = 2.36791e-10, the corrector convergence failed repeatedly or with |h| = hmin.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
2025-01-22 16:00:31.868 - [ERROR] callbacks.on_experiment_error(233): Simulation error: Maximum number of decreased steps occurred at t=8910230.814540884 (final SolverError: 'Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:361:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.'). For a full solution try reducing dt_max (currently, dt_max=300) and/or reducing the size of the time steps or period of the experiment. Set return_solution_if_failed_early=True to return the solution object up to the point where failure occured.

I tried some solutions as suggested by contributors in this issue as well as in other similar issues, for example,

**I have tried by reducing dt_max.
I have tried by changing the period of experiment.
I have tried all modes of Casadisolver.
I have tried IDAKLUSolver.
I have tried CasadiSolver("fast with events")**
I am getting the same error all the time. It will be a great help if you can suggest some other possible solutions to resolve this issue

The python script is as follows for your reference:

import pybamm
import matplotlib.pyplot as plt
import numpy as np
import pandas as pd
pybamm.settings.debug_mode=True

model = pybamm.lithium_ion.DFN(
    {
        "SEI": "solvent-diffusion limited",
        "SEI porosity change": "true",
        "lithium plating": "partially reversible",
        "lithium plating porosity change": "true",  # alias for "SEI porosity change"
        "particle mechanics": ("swelling and cracking", "swelling only"),
        "SEI on cracks": "true",
        "loss of active material": "stress-driven",
        "calculate discharge energy": "true",  # for compatibility with older PyBaMM versions
    }
)
param = pybamm.ParameterValues("OKane2022")

var_pts = {
    "x_n": 5,  # negative electrode
    "x_s": 5,  # separator
    "x_p": 5,  # positive electrode
    "r_n": 30,  # negative particle
    "r_p": 30,  # positive particle
}

#solver = pybamm.IDAKLUSolver()
cycle_number = 1000
exp = pybamm.Experiment(
    [
        "Hold at 4.2 V until C/100 (5 minute period)",
        "Rest for 4 hours (5 minute period)",
        "Discharge at 0.1C until 2.5 V (5 minute period)",  # initial capacity check
        "Charge at 0.3C until 4.2 V (5 minute period)",
        "Hold at 4.2 V until C/100 (5 minute period)",
    ]
    + [
        (
            "Discharge at 2C until 2.5 V",  # ageing cycles
            "Charge at 0.3C until 4.2 V (5 minute period)",
            "Hold at 4.2 V until C/100 (5 minute period)",
        )
    ]
    * cycle_number
    + ["Discharge at 0.1C until 2.5 V (5 minute period)"],  # final capacity check
)

solver = pybamm.CasadiSolver(dt_max=300)

import numpy as np

sim = pybamm.Simulation(model, parameter_values=param, experiment=exp, var_pts=var_pts, solver=solver)
sol=sim.solve()

### Steps to Reproduce

sim = pybamm.Simulation(model, parameter_values=param, experiment=exp, var_pts=var_pts, solver=solver)
sol=sim.solve()

### Relevant log output

```shell

```

rtimms, 2025-01-22:
Try adding some extra events to your model
```
# add extra events
tol = 1e-3
c_e = model.variables["Electrolyte concentration [mol.m-3]"]
eps_s_n = model.variables["Negative electrode active material volume fraction"]
eps_s_p = model.variables["Positive electrode active material volume fraction"]
model.events.extend(
    [
        pybamm.Event("Zero electrolyte cut-off [mol.m-3]", pybamm.min(c_e) - tol),
        pybamm.Event(
            "Minimum negative electrode volume fraction", pybamm.min(eps_s_n) - tol
        ),
        pybamm.Event(
            "Minimum positive electrode volume fraction", pybamm.min(eps_s_p) - to
        ),
    ]
```
this should cause it to fail more "nicely".

You are probably depleting electrolyte or having active material -> 0 somewhere.

I'd also recommend the `IDAKLU` solver as the best choice for this.

gupspriya, 2025-01-22:
Hi, thanks for the quick response. I tried this. Looks like high C-rate at the time of discharging is creating issue.

2025-01-22 16:44:07.886 - [WARNING] callbacks.on_experiment_infeasible_event(252): 

	Experiment is infeasible: 'event: Zero electrolyte cut-off [mol.m-3]' was triggered during 'Discharge at 2C until 2.5 V (5 minute period)'. The returned solution only contains up to step 1 of cycle 569.

gupspriya, 2025-01-23:
Hi, 
I experienced one more error. 
If I am changing lithium plating from 'partially reversible' to 'irreversible', with 0.3C charge and 1C discharge in cycling, I am getting error:
Experiment is infeasible: 'event: Zero electrolyte cut-off [mol.m-3]' was triggered during 'Charge at 0.3C until 4.2 V (5 minute period)'. The returned solution only contains up to step 2 of cycle 79. 

**Can you please suggest possible solutions and reason for it.**

The code is: 

import pybamm
import matplotlib.pyplot as plt
import numpy as np
import pandas as pd
pybamm.settings.debug_mode=True

model = pybamm.lithium_ion.DFN(
    {
        "SEI": "solvent-diffusion limited",
        "SEI porosity change": "true",
        "lithium plating": "irreversible",
        "lithium plating porosity change": "true",  # alias for "SEI porosity change"
        "particle mechanics": ("swelling and cracking", "swelling only"),
        "SEI on cracks": "true",
        "loss of active material": "stress-driven",
        "calculate discharge energy": "true",  # for compatibility with older PyBaMM versions
    }
)
param = pybamm.ParameterValues("OKane2022")
param['Initial concentration in electrolyte [mol.m-3]']=5000

var_pts = {
    "x_n": 5,  # negative electrode
    "x_s": 5,  # separator
    "x_p": 5,  # positive electrode
    "r_n": 30,  # negative particle
    "r_p": 30,  # positive particle
}

tol = 1e-3
c_e = model.variables["Electrolyte concentration [mol.m-3]"]
eps_s_n = model.variables["Negative electrode active material volume fraction"]
eps_s_p = model.variables["Positive electrode active material volume fraction"]
model.events.extend(
    [
        pybamm.Event("Zero electrolyte cut-off [mol.m-3]", pybamm.min(c_e) - tol),
        pybamm.Event(
            "Minimum negative electrode volume fraction", pybamm.min(eps_s_n) - tol
        ),
        pybamm.Event(
            "Minimum positive electrode volume fraction", pybamm.min(eps_s_p) - tol
        ),
    ]
)

solver = pybamm.IDAKLUSolver(rtol=0.000001, atol=1e-06, root_method='casadi', root_tol=1e-06)
cycle_number = 1000
exp = pybamm.Experiment(
    [
        "Hold at 4.2 V until C/100 (5 minute period)",
        "Rest for 4 hours (5 minute period)",
        "Discharge at 0.1C until 2.5 V (5 minute period)",  # initial capacity check
        "Charge at 0.3C until 4.2 V (5 minute period)",
        "Hold at 4.2 V until C/100 (5 minute period)",
    ]
    + [
        (
            "Discharge at 1C until 2.5 V (5 minute period)",  # ageing cycles
            "Charge at 0.3C until 4.2 V (5 minute period)",
            "Hold at 4.2 V until C/100 (5 minute period)",
        )
    ]
    * cycle_number
    + ["Discharge at 0.1C until 2.5 V (5 minute period)"],  # final capacity check
)

sim = pybamm.Simulation(model, parameter_values=param, experiment=exp, var_pts=var_pts, solver=solver)
sol=sim.solve()