pybamm-team/PyBaMM #4796: Separate electrode dimensions

aabills, 2025-01-23:
It's pretty common in practice for the anode and cathode to have different heights and widths (usually by a small margin, admittedly). However, PyBaMM forces them to be the same by not providing separate dimensions for the negative and positive electrodes. This can be compensated by changing the maximum concentration, but this approach is unphysical. Should we change them to be separate?

rtimms, 2025-01-23:
The interpretation here is that height times width gives the overlapping electrode pair area (i.e., it doesn't include the overhang).

aabills, 2025-01-23:
Yes. But that means that the assumption is that the overhang doesn't participate at all, which I doubt is true, though surprisingly, I can't find a 3d study revealing the answer

Rishab87, 2025-02-06:
Hi, I would like to work on this issue, can this be assigned to me?

alighato, 2025-02-07:
I agree with Alec. See for example https://www.sciencedirect.com/science/article/pii/S2590116819300116 p.13.

mpegis, 2025-04-10:
Regardless of whether the overhang is participating in the cell dynamics during practical charge rates, would having the ability to set the dimensions of each unit stack component in the parameter values be beneficial for lumped thermal models, or better estimates of the cell GED?

A common scenario is to have either unpaired anode/cathode or just a single sided coating on the top/bottom of a stacked pouch cell or other form factor.

Is it possible this feature could provide more easy tabulation of the total cell mass and mass% of each component? Or for lumped thermal models, would this provide easier ways to estimate the lumped heat capacity of the cell, even if the unpaired materials did not contribute to the dynamics explicitly?