pybamm-team/PyBaMM #4799: Use InputParameter in pybamm experiment

valentinsulzer, 2025-01-24:
### Description

Allow using input parameters in a pybamm experiment, e.g. 
`pybamm.step.current(pybamm.InputParameter("I_app"), termination="2.5V")`

### Motivation

Firstly, this allows faster sweeping over different conditions

More importantly, we would be able to set up simulations to re-use steps instead of re-discretizing each step. Currently, it does avoid re-discretizing steps if they're exactly the same, but there are some edge cases where it incorrectly thinks steps are different. It also rediscretizes steps even if they have the same structure but different values, which should not be necessary (e.g. "Discharge at 1C until 2.5V" and "Charge at 2C until 4.2V" have the same model structure and should not need to be discretized twice).

### Possible Implementation

Create relevant input parameters when building the simulation experiment, then pass in the appropriate values as inputs at solve time

Rishab87, 2025-02-06:
Hi I would love to work on this issue, can it be assigned to me?

valentinsulzer, 2025-02-07:
Thanks @Rishab87 ! @martinjrobins are there any pitfalls to be aware of with this, with respect to sensitivities for experiments?

Rishab87, 2025-02-08:
@valentinsulzer I've created a PR can you please review it?

PR #4826 (2025-02-21): Added InputParameter support in PyBamm experiments
# Description

Added InputParameter support in experiments, now you can pass them something like this:
```python
current_step = pybamm.step.current(1, input_parameters=pybamm.InputParameter("I_app"), termination="2.5 V")
experiment = pybamm.Experiment(
    input_parameters={"I_app": 1},
    termination="2.5 V",
    operating_conditions=[current_step],
)
```

Fixes #4799 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (includ [... cut]
closed 2025-02-21 by MarcBerliner