pybamm-team/PyBaMM #4808: Another method for OCP with hysteresis

chtamar, 2025-01-30:
### Description

This method has been proposed by Axen et al. https://doi.org/10.1016/j.est.2022.103985, and like the Wycisk approach, it employs an ODE system to track the evolution of the ocp between the empirical lithiation and delithiation branches of the hysteresis. The rate at which the ocp moves back and forth between the hysteresis branches is scaled by the applied current and a decay rate.

### Motivation

Although the Wycisk and Axen methods perform similarly, the proposed implementation does not need to compute the differentiation `self.phase_param.U(sto_surf, T_bulk).diff(sto_surf)`, and it also allows to choose separate decay rates for the lithiation and delithiation branches of the hysteresis. This option can be implemented for the Wycisk approach as well.

Together with the Wycisk approach, this method could be another alternative to choose from in pybamm.

A figure is attached comparing Wycisk and Axen for the silicon's ocp, sweeping over the decay rate (K) and temperature (T), using the _Chen2020_composite_ parametrization and an OCP entropic change for silicon of 2e-3 V/K, for testing purposes. Here the lithiation and delithiation decay rates for Axen take the same value.

[image]

### Possible Implementation

The method can be readily implemented using the module `wycisk_ocp` as starting point.

The method would need the following input parameters:
- The hysteresis branches for lithiation and delithition, `ocp_lith` and `ocp_delith`, respectively.
- The hysteresis decay rate for lithiation `K_lith`
- The hysteresis decay rate for delithiation `K_delith`
- The initial value for the state variable `h_init`

Main features of the method:
- As in the Wycisk approach, the state variable would be `h`, taking values between 0 and 1
- The rhs expression of the state variable would be scaled by the _volumetric interfacial current density_ `i_vol` and the decay rates `K_lith` and `K_delith`
- The magnitude of the OCP hysteresis would be tracked with the variable `H = lith_ref - delith_ref`, where `lith_ref = ocp_lith(sto, T)` and `delith_ref = ocp_delith(sto, T)`
- The resulting OCP is calculated as `ocp_surf = delith_ref + H * h`

The method would be implemented in the new file **_pybamm\models\submodels\interface\open_circuit_potential\axen_ocp.py_**

And the following files would be modified:
- **_pybamm\models\submodels\interface\open_circuit_potential\\\_\_init\_\_.py_**: to make the `axen_ocp` module available in the namespace and import the class `AxenOpenCircuitPotential`
- **_pybamm\models\full_battery_models\base_battery_model.py_** and **_pybamm\models\full_battery_models\lithium_ion\base_lithium_ion_model.py_**: to included the model's ocp option _"Axen"_
- **_pybamm\parameters\lithium_ion_parameters.py_**: to add the new function `U_hysteresis_branch`, which includes the effect of temperature in the hysteresis branches, and to define the parameters `K_lith` and `K_delith`
- **_pybamm\CITATIONS.bib_**: to add the relevant citation

I could contribute with the implementation if you think this is worth including in pybamm.

rtimms, 2025-01-31:
This would be a very welcome addition, please let me know if you need any help with this!

Also see https://github.com/pybamm-team/PyBaMM/issues/4332 for some discussion around refactoring the current implementation of the Wycisk model.

chtamar, 2025-01-31:
That's great @rtimms . I'd be glad to contribute to the project.
I'll test the implementation on the model in #4332, see how it goes and then get the pull request ready.

chtamar, 2025-02-03:
Hi @rtimms, very interesting discussion in #4332.

I'm not sure if this is still relevant but the problem in #4332 seems to be related to the choice of the solver tolerance. The figure below shows the hysteresis state for the Wycisk approach using the Casadi solver with two different tolerances and initial state of `h_init = 0` (pybamm _v25.1.1_). For a tolerance of `1e-3` (used in the model of #4332), the hysteresis state `h` can take values larger than 1 (the bug reported in #4332). But when the tolerance is reduced to `1e-6`, this problem is not observed.

[image]

The Axen method would be a special case of the Wycisk implementation discussed in #4332 when $\gamma$ is a Scalar. However, in contrast to Wycisk, the Axen method skips the average OCP, only the hysteresis branches are required.

I have tested the implementation and it has been pushed to my fork. Should I create a pull request?

rtimms, 2025-02-04:
Great, thanks for the update! Yes, please make a pull request.

ejfdickinson, 2025-02-13:
@rtimms Let me know if you'd like to pick up on last week's in-person conversation on aligning all the hysteresis models.

I think it would probably make sense to package these under a single 'base hysteresis model' or else bring the base treatment of hysteresis within the base OCP model. That would make it much easier to fix #3867 in the general case. I think you or another of the core team would be the best implementer, but I'd be happy to contribute some suggestions on base mathematical formulation.

rtimms, 2025-02-13:
Yes, sounds good. There is an open PR for a new model https://github.com/pybamm-team/PyBaMM/pull/4816, so let’s merge that and then consolidate everything.

ejfdickinson, 2025-02-13:
@rtimms Yes, agreed that that's the right order.

PR #4816 (2025-02-16): Differential method from Axen et al. (2022) for OCP with hysteresis
# Description

- Fixes #4808
- **Motivation**: This method has been proposed by Axen et al. https://doi.org/10.1016/j.est.2022.103985, and like the Wycisk approach, it employs an ODE system to track the evolution of the ocp between the empirical lithiation and delithiation branches of the hysteresis. Although the Wycisk and Axen methods perform similarly, the proposed implementation does not need to compute the differentiation `self.phase_param.U(sto_surf, T_bulk).diff(sto_surf)`, which makes it faster, and it also allows to c [... cut]
closed 2025-02-16 by rtimms