pybamm-team/PyBaMM #4811: Long experiment simulation with DFN

simdieudoboinzemwende, 2025-01-31:
### Description

Hello everyone,

I am a PhD student working on lithium-ion batteries. My project involves the development of a prediction model that integrates a multilayer perceptron (MLP) and an electrochemical model.

I am very interested in working or collaborating with a member of this community who is working on a similar topic.

Furthermore, I am writing this message to ask for your help. I need experimental or simulation data that closely resembles reality to improve my model. I have attempted to run simulations using the DFN model, incorporating various aspects to account for many aging phenomena. However, the problem is that the simulation is extremely slow, taking more than 10 hours for a hundred cycles. If anyone has data or can help me with this issue, I would be extremely grateful.

Additionally, I would like to know if there is any functionality available within the model to accelerate the aging process simulations.

Thank you in advance for your help and promptness.

Best regards,

### Possible Implementation

experiment_long = pybamm.Experiment(
    [
        (
            "Discharge at C/10 until 2.5V",
            "Rest for 10 minutes",
            "Charge at 0.2C until 4.2V",
            "Hold at 4.2V until C/20",
        )+
        (
            "Discharge at C/10 until 2.5V",
            "Rest for 10 minutes",
            "Charge at 1C until 4.2V",
            "Hold at 4.2V until C/20",
            "Rest for 20 minutes",
        )+
        (
            "Discharge at C/10 until 2.5V",
            "Rest for 10 minutes",
            "Charge at 2C until 4.2V",
            "Hold at 4.2V until C/20",
            "Rest for 20 minutes",
        )
    ]
    * 50,
    termination="80% capacity",
)
model_dfn = pybamm.lithium_ion.DFN({'thermal':'lumped',
                                    'SEI': 'solvent-diffusion limited',
                                    'particle mechanics': 'swelling and cracking',
                                    'SEI on cracks':'true',
                                    'lithium plating porosity change':'true',
                                    'lithium plating':'irreversible',
                                    'loss of active material': 'reaction-driven',
                                    'SEI film resistance':'distributed',
                                   })

sim_dfn1=pybamm.Simulation(model_dfn, experiment=experiment_long, parameter_values=param)
sol=sim_dfn1.solve(solver=pybamm.CasadiSolver(mode="safe"))

agriyakhetarpal, 2025-02-01:
Hi @simdieudoboinzemwende, could you please try using the IDAKLU solver? It has been improved significantly with recent versions of PyBaMM for simulating long experiments faster, and it is also available to install separately using `pip install pybammsolvers`. It should be downloaded when you upgrade PyBaMM to a more recent version (25.1, for example).

There is also an example notebook on simulating long experiments that you may read: https://docs.pybamm.org/en/latest/source/examples/notebooks/simulations_and_experiments/simulating-long-experiments.html

Here's the documentation for the IDAKLU solver: https://docs.pybamm.org/en/stable/source/api/solvers/idaklu_solver.html

martinjrobins, 2025-03-06:
Closing this as answered, @simdieudoboinzemwende if you have any further questions feel free to re-open, or use the pybamm discourse group: https://pybamm.discourse.group/

closed 2025-03-06 by martinjrobins