pybamm-team/PyBaMM #4821: [Bug]: unable to install latest pybamm on aarch64

chuckliu1979, 2025-02-07:
### PyBaMM Version

Pybamm 25.1.1

### Python Version

python 3.10

### Describe the bug

since solvers moved to new package pybammsolvers, i can not install pybamm on aarch64, even install pybammsolvers directly:

 pip install pybammsolvers
Defaulting to user installation because normal site-packages is not writeable
Collecting pybammsolvers
  Using cached pybammsolvers-0.1.0.tar.gz (44 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting casadi (from pybammsolvers)
  Using cached casadi-3.6.7-cp310-none-manylinux2014_aarch64.whl.metadata (2.2 kB)
Collecting numpy<2.0 (from pybammsolvers)
  Using cached numpy-1.26.4-cp310-cp310-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (62 kB)
Using cached numpy-1.26.4-cp310-cp310-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (14.2 MB)
Using cached casadi-3.6.7-cp310-none-manylinux2014_aarch64.whl (43.9 MB)
Building wheels for collected packages: pybammsolvers
  Building wheel for pybammsolvers (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for pybammsolvers (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [321 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-aarch64-cpython-310/pybammsolvers
      copying src/pybammsolvers/version.py -> build/lib.linux-aarch64-cpython-310/pybammsolvers
      copying src/pybammsolvers/__init__.py -> build/lib.linux-aarch64-cpython-310/pybammsolvers
      running egg_info
      writing src/pybammsolvers.egg-info/PKG-INFO
      writing dependency_links to src/pybammsolvers.egg-info/dependency_links.txt
      writing requirements to src/pybammsolvers.egg-info/requires.txt
      writing top-level names to src/pybammsolvers.egg-info/top_level.txt
      reading manifest file 'src/pybammsolvers.egg-info/SOURCES.txt'
      adding license file 'LICENSE'
      writing manifest file 'src/pybammsolvers.egg-info/SOURCES.txt'
      /tmp/pip-build-env-fbr81j_m/overlay/lib/python3.10/site-packages/setuptools/command/build_py.py:212: _Warning: Package 'pybammsolvers.idaklu_source' is absent from the `packages` configuration.
      !!
      
              ********************************************************************************
              ############################
              # Package would be ignored #
              ############################
              Python recognizes 'pybammsolvers.idaklu_source' as an importable package[^1],
              but it is absent from setuptools' `packages` configuration.
      
              This leads to an ambiguous overall configuration. If you want to distribute this
              package, please make sure that 'pybammsolvers.idaklu_source' is explicitly added
              to the `packages` configuration field.
      
              Alternatively, you can also rely on setuptools' discovery methods
              (for example by using `find_namespace_packages(...)`/`find_namespace:`
              instead of `find_packages(...)`/`find:`).
      
              You can read more about "package discovery" on setuptools documentation page:
      
              - https://setuptools.pypa.io/en/latest/userguide/package_discovery.html
      
              If you don't want 'pybammsolvers.idaklu_source' to be distributed and are
              already explicitly excluding 'pybammsolvers.idaklu_source' via
[... 274 lines cut ...]
          self.distribution.run_command(command)
        File "/tmp/pip-build-env-fbr81j_m/overlay/lib/python3.10/site-packages/setuptools/dist.py", line 999, in run_command
          super().run_command(command)
        File "/tmp/pip-build-env-fbr81j_m/overlay/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 1002, in run_command
          cmd_obj.run()
        File "<string>", line 138, in run
        File "/usr/local/lib/python3.10/subprocess.py", line 526, in run
          raise CalledProcessError(retcode, process.args,
      subprocess.CalledProcessError: Command '['cmake', '/tmp/pip-install-ml11b504/pybammsolvers_a88d40bc6d0547688960ec5d0c66dd00', '-DCMAKE_BUILD_TYPE=RELEASE', '-DPYTHON_EXECUTABLE=/usr/local/bin/python3.10', '-DUSE_PYTHON_CASADI=TRUE', '-DPYBAMM_IDAKLU_EXPR_CASADI=ON', '-DPYBAMM_IDAKLU_EXPR_IREE=OFF', '-DSuiteSparse_ROOT=/tmp/pip-install-ml11b504/pybammsolvers_a88d40bc6d0547688960ec5d0c66dd00/.idaklu', '-DSUNDIALS_ROOT=/tmp/pip-install-ml11b504/pybammsolvers_a88d40bc6d0547688960ec5d0c66dd00/.idaklu']' returned non-zero exit status 1.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pybammsolvers
Failed to build pybammsolvers
ERROR: Failed to build installable wheels for some pyproject.toml based projects (pybammsolvers)

### Steps to Reproduce

pip install pybammsolvers

### Relevant log output

```shell

```

kratman, 2025-02-07:
@chuckliu1979 We are working on linux arm64 support at the moment. This will be fixed in the near future, we have two different work packages trying to resolve this issue

kratman, 2025-02-12:
This will be covered by this ticket: https://github.com/pybamm-team/pybammsolvers/issues/19

I am working on the issues now

agriyakhetarpal, 2025-02-12:
I think it makes sense to keep it open for visibility, for now, as this affects both PyBaMM and `pybammsolvers`. Your PR https://github.com/pybamm-team/pybammsolvers/pull/28 (or a newer one after that) could reference this issue to close it as both repositories are under the same organisation.