pybamm-team/PyBaMM #4833: Deprecate number of electrodes in parallel to make a cell

aabills, 2025-02-10:
The "number of electrodes in parallel to make a cell" (potentially paraphrased) parameter is confusing and not always used properly. For example, it is not used to calculate the area of the current collector: https://github.com/pybamm-team/PyBaMM/blob/74192198058fd914afd9130dbc201fcc673fab20/src/pybamm/parameters/geometric_parameters.py#L44, which is used to calculate the "total lithium in the electrolyte": https://github.com/pybamm-team/PyBaMM/blob/74192198058fd914afd9130dbc201fcc673fab20/src/pybamm/parameters/lithium_ion_parameters.py#L116

We could go through and try and find and fix all of these, or we could just deprecate this parameter. I think we should do the latter.

valentinsulzer, 2025-02-15:
Happy to remove this in favor of just scaling the electrode area properly. How about the number of cells in series? It only really applies to lead-acid batteries (to get to 12V when the OCV is ~2V)

aabills, 2025-02-16:
Should we just switch to "Electrode area [m2]" instead?

ejfdickinson, 2025-02-17:
I think that here the best development would be disentangling the parameterisation of the electrical model (relation of cell current to 1D macroscopic current density), and any parameterisation of the actual layout of the cell, for thermal modelling or other purposes.

Even if you can count your electrodes and their coatings, you need to make some assumption about what the 'out-of-plane' area of the SPM/SPMe/DFN should be, because negative electrode, positive electrode and separator all have different areas and while basically only the facing area counts, there can be edge effects.. So I'd recommend a simple `"Electrode area [m2]"` for intensive -> extensive scaling purposes, such that `Cell current [A] = Electrode geometric current density [A.m-2] * Electrode area [m2]`, and have any other more complicated parameters (height, width, number of layers, shape, funny stuff) exclusively for more complicated purposes.

valentinsulzer, 2025-02-17:
> It's not obvious to new PyBaMM users how to account for multiple layers in a cell. For example, Chen et al. and the papers based on it (O'Kane et al., O'Regan et al.) handle the double-coated electrode by doubling the electrode width, but other papers use different conventions.

I think removing the "number of electrodes in parallel" will help with this, as we can make it clear that the electrode width/area corresponds to a single anode/separator/cathode layer

> So I'd recommend a simple "Electrode area [m2]" for intensive -> extensive scaling purposes, such that Cell current [A] = Electrode geometric current density [A.m-2] * Electrode area [m2], and have [...]

Sounds like a good idea. The challenge is what if the user has a 3D pouch cell model requiring inputting height and width - in that case should they also specify the electrode area (and be responsible for making sure it's compatible with the height and width), or would PyBaMM calculate it for them based on height and width they provide? And how does this interact with the thermal model? Perhaps a solution is just that different form factors require different parameters.

ejfdickinson, 2025-02-17:
@valentinsulzer My opinion would be to make the user responsible in all cases, because the role of the parameter value in PyBaMM is to define the coefficients for the model equations, not to oblige a specific physical interpretation.

For example, even when specifying "height and width", what do we mean? The electrode sheet dimensions (current collector)? The coated dimensions? They're not co-terminous. We have opened pouch cells with slightly chamfered corners on the electrodes so the coated area isn't even a rectangle if you're being pedantic.

In any case you have some set of actually measurable dimensions, which are difficult to generalise from form factor to form factor (or even cell design to cell design), and then there's some set of needed coefficients in the model equations. For the thermal model these needed coefficients are different and more numerous, and trace back to more materials besides active materials, but the connection to geometry remains a little distant because what we ultimately need are total thermal masses and effective conductivities.

valentinsulzer, 2025-02-17:
Yes, I think that makes sense. So it sounds like what we need is this:
- Use electrode area for standard P2D
- Use height and width for higher dimensional models
- Always specify the thermal volume separately (should we just deprecate the "per layer" thermal models entirely? So the user just provides the heat capacity and heat transfer coefficient, they can calculate those themselves from the layers if they want?)
- Have a page in the docs clearly explaining this, what "electrode area" means (double the double-coated area), etc

vidipsingh, 2025-02-25:
Hi all,

Thanks for the detailed discussion on this! I agree that deprecating the "Number of electrodes in parallel" parameter in favor of using "Electrode area [m²]" sounds like a good way forward. It simplifies the model and removes potential confusion, especially for newer PyBaMM users.

I also think that handling multiple layers by scaling the electrode width/area, as you mentioned, is a practical approach. For higher-dimensional models, like 3D pouch cells, I agree that providing both height and width makes sense for user flexibility.

Just to clarify the next steps and best implementation:

- Should we go ahead and replace the "Number of electrodes in parallel" with "Electrode area [m²]" across all models now?
- For thermal models, would it be best for users to specify thermal properties directly (heat capacity, heat transfer coefficient), and we remove the "per layer" thermal models as suggested?
- Finally, would it make sense to add documentation on these changes, especially explaining how "Electrode area" works in various use cases (e.g., double-coated electrodes, etc.)?

As I'm still new to PyBaMM, any guidance from all of you would be very helpful for me in implementing these changes effectively.

Looking forward to hearing your thoughts on this, and happy to help implement these changes.

PR #5138 (2025-07-29): Make A_cc the full current collector area
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context.
Fixes #4833 

Fixes # (issue)

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #)

# Important checks:

Please confirm the following before marking the PR as ready for review:
- No style issues: `nox -s pre-commit`
- All tests pass: `nox -s tests`
- The documentation builds: `nox -s doctests`
 [... cut]
closed 2025-07-29 by aabills