pybamm-team/PyBaMM #4868: [Bug]: SolverError when simulating a half-cell and setting initial_soc

d-cogswell, 2025-02-21:
### PyBaMM Version

25.1.1

### Python Version

3.12.4

### Describe the bug

`working electrode` and `initial_soc` both work independently, but cause a `SolverError` when used together.

### Steps to Reproduce

Here is a working example:
```
import pybamm

experiment = pybamm.Experiment(["Discharge at 1C until 3V"])
parameter_values = pybamm.ParameterValues("Xu2019")
parameter_values.update(
    pybamm.ParameterValues("Chen2020"), check_already_exists=False
)

# This works
model = pybamm.lithium_ion.SPM()
sim = pybamm.Simulation(model, parameter_values=parameter_values, experiment=experiment)
sol = sim.solve(initial_soc=0.8)

# This works
model = pybamm.lithium_ion.SPM(options={"working electrode": "positive"})
sim = pybamm.Simulation(model, parameter_values=parameter_values, experiment=experiment)
sol = sim.solve()

# This fails
model = pybamm.lithium_ion.SPM(options={"working electrode": "positive"})
sim = pybamm.Simulation(model, parameter_values=parameter_values, experiment=experiment)
sol = sim.solve(initial_soc=0.8)
```

### Relevant log output

```shell
pybamm.expression_tree.exceptions.SolverError: Could not find acceptable solution: solver terminated successfully, but maximum solution error (0.00027437641369143506) above tolerance (1e-06)
```

d-cogswell, 2025-02-24:
Switching from `scipy.optimize.root()` to `scipy.optimize.minimize()` as the algebraic solver seems to fix the issue.

martinjrobins, 2025-02-26:
Are you happy with the fix of changing to a different algebraic solver @d-cogswell? If so I'll close this issue.

agriyakhetarpal, 2025-02-26:
I assume #4873 is supposed to close this, but it's a draft and isn't linked to this issue yet.

d-cogswell, 2025-03-03:
@agriyakhetarpal that's correct. Thanks for linking.

closed 2025-03-05 by kratman