pybamm-team/PyBaMM #4888: Examine usage of `repo-review` tools as pre-commit hooks

agriyakhetarpal, 2025-03-02:
As a follow-up to #3489 and #4887

Saransh-cpp, 2025-03-03:
Just to be clear, I think this issue is independent of the issues mentioned above, and ideally, the hook should be added to the repository as soon as possible (priority is still low). We can ignore the checks that are failing right now, and then slowly _un-ignore_ them as part of #3489.

agriyakhetarpal, 2025-03-03:
Sounds good to me, thanks for clarifying – I was under the impression that you want existing tests to pass first (or with the process of adding it).

medha-14, 2025-03-07:
I benchmarked this, and the results are as follows:

 Without the hook:
[image]

With the hook:

[image]

[image]

Saransh-cpp, 2025-03-07:
Could you please ignore the failed checks in pyproject.toml for the time being and make this hook pass?

agriyakhetarpal, 2025-03-07:
The speed is reasonable, too. We should be fine with adding it, I guess.

Saransh-cpp, 2025-03-07:
Ah, I did not think about the speed at all. It should not matter if it takes a few more seconds!

PR #4897 (2025-03-08): Added `repo-review` pre-commit hook
# Description

Fixes #4888 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #)

# Important checks:

Please confirm the following before marking the PR as ready for review:
- No style issues: `nox -s pre-commit`
- All tests pass: `nox -s tests`
- The documentation builds: `nox -s doctests`
- Code is commented for hard-to-understand areas
- Tests added that prove fix is effective or that feature works
closed 2025-03-08 by Saransh-cpp