pybamm-team/PyBaMM #4896: Docker workflow failing because of IDAKLU solver decoupling

santacodes, 2025-03-07:
### PyBaMM Version

25.1.1

### Python Version

3.11

### Describe the bug

The docker workflow has been failing because of decoupled `IDAKLU` solver. I think it would be better to refactor the `Dockerfile` completely as `pybammsolvers` is an independent dependency now and can be installed through `pip` instead of compiling it within the `Dockerfile` itself. This would make the `Dockerfile` much simpler with less dependencies to install and overall less build steps.

### Steps to Reproduce

Reference workflow URL - https://github.com/pybamm-team/PyBaMM/actions/runs/13685113885/job/38266614422

### Relevant log output

```shell
#35 57.40   × Failed to build `pybammsolvers==0.1.0`
#35 57.40   ├─▶ The build backend returned an error
#35 57.41   ╰─▶ Call to `setuptools.build_meta.build_wheel` failed (exit status: 1)
#35 57.41 
#35 57.41       [stdout]
#35 57.41       running bdist_wheel
#35 57.41       running build
#35 57.41       running build_py
#35 57.41       creating build/lib.linux-aarch64-cpython-312/pybammsolvers
#35 57.41       copying src/pybammsolvers/version.py ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers
#35 57.41       copying src/pybammsolvers/__init__.py ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers
#35 57.41       running egg_info
#35 57.41       writing src/pybammsolvers.egg-info/PKG-INFO
#35 57.41       writing dependency_links to
#35 57.41       src/pybammsolvers.egg-info/dependency_links.txt
#35 57.41       writing requirements to src/pybammsolvers.egg-info/requires.txt
#35 57.41       writing top-level names to src/pybammsolvers.egg-info/top_level.txt
#35 57.41       reading manifest file 'src/pybammsolvers.egg-info/SOURCES.txt'
#35 57.41       adding license file 'LICENSE'
#35 57.41       writing manifest file 'src/pybammsolvers.egg-info/SOURCES.txt'
#35 57.41       copying src/pybammsolvers/idaklu.cpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers
#35 57.41       creating build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
#35 57.41       copying src/pybammsolvers/idaklu_source/IDAKLUSolver.cpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
#35 57.41       copying src/pybammsolvers/idaklu_source/IDAKLUSolver.hpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
#35 57.41       copying src/pybammsolvers/idaklu_source/IDAKLUSolverGroup.cpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
#35 57.41       copying src/pybammsolvers/idaklu_source/IDAKLUSolverGroup.hpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
#35 57.41       copying src/pybammsolvers/idaklu_source/IDAKLUSolverOpenMP.hpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
#35 57.41       copying src/pybammsolvers/idaklu_source/IDAKLUSolverOpenMP.inl ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
#35 57.41       copying src/pybammsolvers/idaklu_source/IDAKLUSolverOpenMP_solvers.cpp
#35 57.41       -> build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
#35 57.41       copying src/pybammsolvers/idaklu_source/IDAKLUSolverOpenMP_solvers.hpp
#35 57.41       -> build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
#35 57.41       copying src/pybammsolvers/idaklu_source/IdakluJax.cpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
#35 57.41       copying src/pybammsolvers/idaklu_source/IdakluJax.hpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
#35 57.41       copying src/pybammsolvers/idaklu_source/Options.cpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
#35 57.41       copying src/pybammsolvers/idaklu_source/Options.hpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
#35 57.41       copying src/pybammsolvers/idaklu_source/Solution.cpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source
[... 51 lines cut ...]
#35 57.41       src/pybammsolvers/idaklu_source/Expressions/IREE/IREEFunctions.cpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source/Expressions/IREE
#35 57.41       copying
#35 57.41       src/pybammsolvers/idaklu_source/Expressions/IREE/IREEFunctions.hpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source/Expressions/IREE
#35 57.41       copying
#35 57.41       src/pybammsolvers/idaklu_source/Expressions/IREE/ModuleParser.cpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source/Expressions/IREE
#35 57.41       copying
#35 57.41       src/pybammsolvers/idaklu_source/Expressions/IREE/ModuleParser.hpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source/Expressions/IREE
#35 57.41       copying src/pybammsolvers/idaklu_source/Expressions/IREE/iree_jit.cpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source/Expressions/IREE
#35 57.41       copying src/pybammsolvers/idaklu_source/Expressions/IREE/iree_jit.hpp ->
#35 57.41       build/lib.linux-aarch64-cpython-312/pybammsolvers/idaklu_source/Expressions/IREE
#35 57.41       running build_ext
#35 57.41       ---------- Running CMake for IDAKLU solver
#35 57.41       ----------------------------------------
#35 57.41 
#35 57.41       [stderr]
#35 57.41       /home/pybamm/.cache/uv/builds-v0/.tmpGLrdF0/lib/python3.12/site-packages/setuptools/command/build_py.py:212:
#35 57.41       _Warning: Package 'pybammsolvers.idaklu_source' is absent from the
#35 57.41       `packages` configuration.
#35 57.41       !!
```

agriyakhetarpal, 2025-03-07:
Thanks, @santacodes. We should also remove the arm64 Docker images as they won't work without https://github.com/pybamm-team/pybammsolvers/issues/19 and #4821 being resolved.

medha-14, 2025-03-07:
Hey @santacodes, could you assign this one to me ?

RohitP2005, 2025-03-13:
Is this still being worked on @medha-14 ?

medha-14, 2025-03-13:
Yes

agriyakhetarpal, 2025-03-29:
Hi @medha-14, do you need any help with this?

Just in case it's unclear, the only change needed at the moment to fix this issue is to remove `linux/arm64` from the `platforms:` input being passed to `docker/build-push-action`:

https://github.com/pybamm-team/PyBaMM/blob/df0ce6876911ed434cd83455ddf81d6813848dab/.github/workflows/docker.yml#L32-L39