pybamm-team/PyBaMM #4902: how to set multiple RC parameters for Thevenin Model

citizen-z1, 2025-03-09:
I noticed in the official documentation that the "number of rc elements" parameter of the Thevenin Model seems to indicate that this model can be configured as a second-order model.
   However, I'm unsure how to set multiple RC parameters. I printed out the parameter list for this model, and it only includes R1 and C1. How can I set the values for R2 and C2？

[image]

[image]

valentinsulzer, 2025-03-11:
you pass options to the model like this: `Thevenin({"number of rc elements": 2})`

citizen-z1, 2025-03-11:
> 您将选项传递给模型，如下所示：`Thevenin({"number of rc elements": 2})`

I know this parameter and have tried it, but when I print the parameter list of the model, there are still no R2 and C2 parameters

[image]

closed 2025-06-18 by kratman