pybamm-team/PyBaMM #4933: Run `zizmor` over all GitHub Actions workflow files

agriyakhetarpal, 2025-03-25:
https://github.com/woodruffw/zizmor is a static analysis tool for GitHub Actions that we should use to evaluate and improve the security of our repository.

Here's the documentation: https://woodruffw.github.io/zizmor/

- [ ] Run zizmor over all workflow files with the `--pedantic` option
- [ ] Address all issues raised by it
- [ ] Add it as a pre-commit hook

JC230903, 2025-03-27:
Hi @agriyakhetarpal, I’d like to work on this issue! 
I’ll start by running Zizmor on my local system to check the findings and understand the necessary fixes. Once I have a clearer picture, I’ll address the issues and look into integrating Zizmor as a pre-commit hook.

I’ll update here once I’ve analyzed the results. Let me know if there are any specific concerns I should keep in mind while working on this. Looking forward to contributing!

agriyakhetarpal, 2025-03-27:
That sounds good to me, thanks! It's definitely useful to explore the tool locally and get an idea of what it's about. Looking forward to your results!

JC230903, 2025-03-27:
Hi @agriyakhetarpal,

I’ve run Zizmor on all the GitHub workflow files and set up a pre-commit hook. i have found that the audit has  identified several key issues:

- [x] Run zizmor over all workflow files with the --pedantic option

- [ ] Address all issues raised by it

- [x] Add it as a pre-commit hook

High Confidence:
Template Injection: Lack of input validation in run_benchmarks_over_history.yml.
Excessive Permissions: Some workflows use read-all permissions.

Medium Confidence:
Broad Permissions: Workflows are using overly general permissions.
Default Permissions: Permissions should be explicitly defined.
Low Confidence:
Credential Persistence: GitHub Actions artifacts may persist credentials.
Cache Poisoning: Docker builds are at risk of cache poisoning.

Main Causes:
Missing input validation.
Broad or undefined permissions.
Not setting persist-credentials: false in checkout actions.
Default permissions and improper caching in Docker.
let me know if you recommend any key areas for me to work on!

agriyakhetarpal, 2025-03-27:
Thanks, @JC230903. I think it would be good to address all the issues noted at once and add the hook to `pre-commit-config.yaml` in the same PR. Please feel free to open a PR, and I can help out if you get stuck somewhere.

JC230903, 2025-03-27:
@agriyakhetarpal 

That sounds good—I’ll address all the issues  and update pre-commit-config.yaml in the same PR. Before I start, do you have any specific recommendations or things I should keep in mind while making these changes?

I’ll open a PR soon and will reach out if I run into any issues.

agriyakhetarpal, 2025-03-27:
> Before I start, do you have any specific recommendations or things I should keep in mind while making these changes?

Not at the moment, thanks!

You may refer to https://github.com/scikit-image/scikit-image/pull/7662 to see how other projects are doing it.

kratman, 2025-03-27:
FYI: https://github.com/pybamm-team/PyBaMM/security/code-scanning

Dependabot already does a bunch of these checks

agriyakhetarpal, 2025-03-27:
Yes, I'm aware, @kratman. The idea is that Dependabot doesn't run ahead of time and GitHub doesn't seem to consider this page to be prominent enough to surface it more often. Having a pre-commit hook solves this in some sort because it prevents improper workflows from being pushed.

PR #4935 (2025-03-31): Fix zizmor security issues over Github Actions Workflows
# Description
This PR addresses multiple security and best practice issues identified by Zizmor in our GitHub Actions workflows. The changes improve the overall security posture of our CI/CD pipeline by eliminating hardcoded credentials, mitigating injection vulnerabilities, and implementing GitHub Actions best practices.
Fixes #4933

- [x] Run zizmor over all workflow files with the --pedantic option
- [x] Address all issues raised by it
- [x] Add it as a pre-commit hoo

Zizmor identified 51 findings across our workflows (8 high, 42 med [... cut]
closed 2025-03-31 by agriyakhetarpal