pybamm-team/PyBaMM #4938: Benchmarks failing

Saransh-cpp, 2025-03-30:
All the benchmarks (except the [validation](https://github.com/pybamm-team/pybamm-validation/issues/8) ones, because they are commented out) are failing ([periodic](https://github.com/pybamm-team/PyBaMM/actions/workflows/periodic_benchmarks.yml), [on_push](https://github.com/pybamm-team/PyBaMM/actions/workflows/benchmark_on_push.yml)). We need a better way (maybe a badge in the README) to display the failures as they have been failing for quite some time now and no one has noticed it.

I haven't looked into the failure, but I am guessing that they are still using some old PyBaMM code.

agriyakhetarpal, 2025-03-30:
So it looks like the benchmarks are alright, they are just failing because of this issue: https://github.com/pympler/pympler/issues/151 caused by a specific benchmark that didn't occur before. We can disable the relevant benchmark if it helps.

agriyakhetarpal, 2025-03-30:
So it looks like the benchmarks are alright, they are just failing because of this issue: https://github.com/pympler/pympler/issues/151, caused by a specific benchmark that didn't occur before for earlier versions of Pympler. We can disable the relevant benchmark if it helps.

Saransh-cpp, 2025-03-30:
Sounds good!