pybamm-team/PyBaMM #4949: [Bug]: Minor patch release for Casadi

BradyPlanden, 2025-04-02:
### PyBaMM Version

25.1

### Python Version

3.12

### Describe the bug

Hi All - is there a plan/timeline to release a patch for the pinned casadi dependency? At the moment, new installations of PyBaMM are broken due to this dependency.

### Steps to Reproduce

Install pybamm and run an example

### Relevant log output

```shell

```

agriyakhetarpal, 2025-04-02:
Hi @BradyPlanden! Yes, @kratman is working on a release. Though, I guess it would be nice to get a 25.1.post1 release with just #4945 on top of `main`.

kratman, 2025-04-02:
Yeah v25.4.0 should be released today. I am just waiting on 1 PR that Rob wants to get in this release. Should be done in a couple hours

agriyakhetarpal, 2025-04-02:
Yes, that should be fine; thanks. I would still suggest creating another post1 release for 25.1 with just this metadata change, as users might need more time to adapt their code for 25.4 and 25.1 would still be broken.

agriyakhetarpal, 2025-04-02:
Yes, that should be fine; thanks. I would still suggest creating another post1 release for 25.1 with just this metadata change, as users might need more time to adapt their code for 25.4, and 25.1 would still be broken.

kratman, 2025-04-02:
I mean all of our old releases without an exact pin for casadi are broken. At some point we just have to cut our losses and do a fresh release

closed 2025-04-03 by BradyPlanden