pybamm-team/PyBaMM #4970: refactor: always return_jacp_stacked

martinjrobins, 2025-04-10:
The `process` function in `base_solver.py` is used to either take the sensitivities wrt each parameter in turn, or take the sensitivities wrt the parameter vector (if true). The later is neccessary to calculate online sensitivities if `output_variables` is set.

This refactor proposes to always have return_jacp_stacked=True, in order to simplify the code that needs to deal with each case individually.

Might have to do this anyway for https://github.com/pybamm-team/PyBaMM/issues/4547, but I'll just put this issue in place for a reminder to do this.

martinjrobins, 2025-04-10:
could have the python and jax sensitivity functions also return the same as return_jacp_stacked=True so that everything is aligned

RohitP2005, 2025-04-11:
@martinjrobins I'd like to work on this issue if it's available to be picked up. Let me know if that's okay!