pybamm-team/PyBaMM #4988: [Bug]: Exponential1DSubMesh gives bad results

mleot, 2025-04-22:
### PyBaMM Version

25.4.0

### Python Version

3.11.11

### Describe the bug

When running a simple DFN model and using the Exponential1DSubmesh, the results end up looking very strange. 

[image]

I assume this is not just a plotting issue, though I am not sure. 

### Steps to Reproduce

```
import pybamm

model = pybamm.lithium_ion.DFN()
parameter_values = model.default_parameter_values
parameter_values.process_model(model)

geometry = model.default_geometry
parameter_values.process_geometry(geometry)

submesh_types = model.default_submesh_types
submesh_types = {}
submesh_types.update({
    "negative electrode": pybamm.Exponential1DSubMesh,
    "negative particle": pybamm.Exponential1DSubMesh,
    "separator": pybamm.Exponential1DSubMesh,
    "positive electrode": pybamm.Exponential1DSubMesh,
    "positive particle": pybamm.Exponential1DSubMesh,
    "current collector": pybamm.SubMesh0D,
})
mesh = pybamm.Mesh(geometry, submesh_types, model.default_var_pts)
disc = pybamm.Discretisation(mesh, model.default_spatial_methods,check_model=True)

disc.process_model(model)

var_pts = {
    "x_n": 30,  # negative electrode
    "x_s": 20,  # separator
    "x_p": 30,  # positive electrode
    "r_n": 20,  # negative particle
    "r_p": 20,  # positive particle
}

sim = pybamm.Simulation(
    model=model,
    submesh_types=submesh_types,
    var_pts=var_pts,
    solver = pybamm.IDAKLUSolver()
)

sim.solve([0,3600])
```

### Relevant log output

```shell

```

rtimms, 2025-04-23:
This is a bug in the mesh. Looks like it assumes the left most point in the mesh in zero

```python3
import pybamm
import matplotlib.pyplot as plt

model = pybamm.lithium_ion.DFN()

submesh_types = model.default_submesh_types
submesh_types.update(
    {
        "negative particle": pybamm.MeshGenerator(
            pybamm.Exponential1DSubMesh, submesh_params={"side": "right"}
        ),
        "positive particle": pybamm.MeshGenerator(
            pybamm.Exponential1DSubMesh, submesh_params={"side": "right"}
        ),
        "current collector": pybamm.SubMesh0D,
    }
)
var_pts = {
    "x_n": 30,  # negative electrode
    "x_s": 20,  # separator
    "x_p": 30,  # positive electrode
    "r_n": 20,  # negative particle
    "r_p": 20,  # positive particle
}
sols = []
domains = ["negative electrode", "separator", "positive electrode"]
sides = ["symmetric", "left", "right"]
labels = []
fig, ax = plt.subplots()
for domain in domains:
    for side in sides:
        try:
            this_submesh_types = submesh_types.copy()
            this_submesh_types[domain] = pybamm.MeshGenerator(
                pybamm.Exponential1DSubMesh, submesh_params={"side": "symmetric"}
            )
            sim = pybamm.Simulation(
                model=model,
                submesh_types=this_submesh_types,
                var_pts=var_pts,
                solver=pybamm.IDAKLUSolver(),
            )
            sol = sim.solve([0, 100])
            sols.append(sol)
            labels.append(f"{domain} {side}")
            print(
                f"Solved {domain}, {side}",
                "t_end =",
                sol["Time [s]"].entries[-1],
                "V_end =",
                sol["Voltage [V]"].entries[-1],
            )
            ax.plot(sol["Time [s]"].entries, sol["Voltage [V]"].entries, "o", label=f"{domain} {side}", alpha=0.5)
        except pybamm.SolverError as e:
            print(f"Solver error for {domain}, {side}: {e}")
ax.legend()
ax.set_xlabel("Time [s]")
ax.set_ylabel("Voltage [V]")
```

[image]

rtimms, 2025-04-23:
Ok, it starts at one but calculates the edges incorrectly:

```python3
r = pybamm.SpatialVariable("r", domain=["negative particle"], coord_sys="spherical polar")
submesh_params = {"side": "symmetric", "stretch": 1.15}
submesh_types = {
    "negative particle": pybamm.MeshGenerator(
        pybamm.Exponential1DSubMesh, submesh_params
    )
}
var_pts = {r: 10}
geometry_0_1 = {
    "negative particle": {r: {"min": pybamm.Scalar(0), "max": pybamm.Scalar(1)}}
}
geometry_1_2 = {
    "negative particle": {r: {"min": pybamm.Scalar(1), "max": pybamm.Scalar(2)}}
}
mesh_0_1 = pybamm.Mesh(geometry_0_1, submesh_types, var_pts)
mesh_1_2 = pybamm.Mesh(geometry_1_2, submesh_types, var_pts)

print(mesh_0_1["negative particle"].edges)
print(mesh_1_2["negative particle"].edges)
```

gives 
```
[0.         0.02823507 0.05991124 0.09544789 0.13531552 0.5
 0.86468448 0.90455211 0.94008876 0.97176493 1.        ]
[1.  1.  1.  1.  1.  1.5 2.  2.  2.  2.  2. ]
```

rtimms, 2025-04-23:
output after fix

[image]

PR #4989 (2025-04-23): Fix exponential mesh bug
# Description

Fixes a bug with `Exponential1DSubMesh` when the left edge is non-zero.

Fixes #4988 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #)

# Important checks:

Please confirm the following before marking the PR as ready for review:
- No style issues: `nox -s pre-commit`
- All tests pass: `nox -s tests`
- The documentation builds: `nox -s doctests`
- Code is commented for hard-to-understand areas
- Tests added that prove fix i [... cut]
closed 2025-04-23 by kratman