pybamm-team/PyBaMM #5019: [Bug]: TypeError: must be real number, not BoundaryValue when using a callable controller in CustomStepExplicit

leigedove, 2025-05-16:
### PyBaMM Version

25.1.1

### Python Version

3.10.16

### Describe the bug

Hi PyBaMM team, thanks for your fantastic work on this project.

I’m experimenting with custom control strategies using CustomStepExplicit, where the current is determined by a user-defined callable function that depends on variables like voltage.

However, I'm encountering the following error when running the simulation:
TypeError: must be real number, not BoundaryValue

do not yield plain Python floats, but instead return symbolic objects (likely of type BoundaryValue or similar), which cannot be directly consumed by downstream numerical libraries.

Are there any solutions for these problems?

Thank you!

Best, 
Lei Ge

### Steps to Reproduce

```
class VoltageBasedController(nn.Module):
    def __init__(self):
        super(VoltageBasedController, self).__init__()
        self.net = nn.Sequential(
            nn.Linear(1, 16),
            nn.ReLU(),
            nn.Linear(16, 1)
        )

    def forward(self, x):
        return self.net(x)

# Instantiate and freeze the model
model = VoltageBasedController()
model.eval()

# External controller function
def voltage_controller(variables):
    V = variables["Voltage [V]"]  # Expected to be a float
    V_tensor = torch.tensor([[V]], dtype=torch.float32)
    I_pred = model(V_tensor).item()
    return I_pred

# Define PyBaMM model
battery_model = pybamm.lithium_ion.DFN()

# Define experiment with external control on voltage
step = pybamm.step.CustomStepExplicit(
    voltage_controller,
    duration=600,
    termination="4.2V",
    direction="charge",
    external_variables=["Voltage [V]"]  # Ensure PyBaMM passes real values
)

# Run simulation
sim = pybamm.Simulation(battery_model, experiment=step)
solution = sim.solve()
sim.plot()
```

### Relevant log output

```shell

```

kratman, 2025-05-19:
@leigedove Can you post a minimum working example and the error?

I would also recommend upgrading to pybamm 25.4.2

leigedove, 2025-05-20:
Thank you for you reply! I've upgrade to the latest pybamm version, and it still doesn't work. Following is a simple example: 

```
import torch
import torch.nn as nn
import pybamm

def nn_controller(variables):

    class SimpleChargingNN(nn.Module):
        def init(self):
            super(SimpleChargingNN, self).init()
            self.net = nn.Sequential(
            nn.Linear(2, 8),
            nn.ReLU(),
            nn.Linear(8, 1)
            )
        def forward(self, x):
            return self.net(x)

    model = SimpleChargingNN()
    model.eval() 
    
    V = variables["Voltage [V]"]
    T = variables.get("Cell temperature [K]", variables.get("Ambient temperature [K]", 298.15))
    inp = torch.tensor([[V, T]], dtype=torch.float32)
    I_pred = model(inp).item()
    return I_pred

battery_model = pybamm.lithium_ion.DFN()

step = pybamm.step.CustomStepExplicit(
nn_controller, duration=600, termination="4.2V", direction="charge"
)

sim = pybamm.Simulation(battery_model, experiment=step)
solution = sim.solve()
sim.plot()
```

Thanks a lot for your help!
Best, 
Lei Ge